hybridgroup/gobot #171: How to contribute to the Gobot project?

mwskz8, 2015-01-11:
Hi,

Do you have any info about how people can contrbute to the Gobot project?  Thanks.  

Thanks
mwskz

zankich, 2015-01-11:
Hi @mwskz8 

A great way to contribute to Gobot is by adding documentation. Our documentation is still a work in progress and kinda sparse in some places, so any help there would be very much appreciated! Another easy way to contribute is by adding tests. 

When a person asks for a new feature we don't currently support, I mark it with the `enhancement` tag, so those are pretty much open if somebody wants to jump in and help get that implemented.

We often get requests for new platform support that we don't currently have, and I want to make it easy for people to link their platform implementations from the main Gobot readme/website, I need to write something about that in our readme. I've also been playing around with the idea of having something like a `gobot-contrib` repo or org where people can host their platform implementations.

As far as contribution guidelines, we have a section at the bottom of the [readme](https://github.com/hybridgroup/gobot#contributing) which details that.

deadprogram, 2015-03-27:
@mwskz8 hopefully this answers your questions. We will reopen if needed. Thanks!

closed 2015-03-27 by deadprogram