hybridgroup/gobot #172: Servos: setting minimum and maximum "pulse width" - Arduino

pipejesus, 2015-01-15:
Using the Arduino libraries i can set the maximum and minimum pulse width for my servo so that i can calibrate it. Is it possible with Gobot ? What i mean is this : http://arduino.cc/en/Reference/ServoAttach

zankich, 2015-01-15:
@pipejesus at the moment no, we don't have that capability, however I'm currently working on a refactor of our firmata implementation and based off of the [servo](https://github.com/firmata/protocol/blob/master/servos.md) protocol document, it should be easy to add.

pipejesus, 2015-01-15:
Superb! Great project, thank you Adrian for bringing Gobot to Earth :-)

losinggeneration, 2016-07-17:
It looks like this might be implemented (at least on the client library.) https://github.com/hybridgroup/gobot/blob/dev/platforms/firmata/client/client.go#L217 However, it looks like the bytes are sent as max,min instead of min,max unless I'm misunderstanding the code and/or protocol doc.

deadprogram, 2016-07-19:
Hi @losinggeneration looks to me like you are absolutely correct. Would you be willing to submit a PR with that fix?

losinggeneration, 2016-07-20:
I'll pick up a servo tomorrow to verify it works and then get a PR in for it.

deadprogram, 2016-11-05:
Seems like this was addressed already. Please reopen if needed.

closed 2016-11-05 by deadprogram