hybridgroup/gobot #173: Running firmata_blink on Arduino fails on second attempt.

dkinzer, 2015-02-04:
If I control C out of a running process and retry then the new process hangs unless I unplug and replug the USB port from the Arduino device.

```

(examples)(master)$./firmata_blink 
2015/02/04 09:51:06 Initializing Robot bot ...
2015/02/04 09:51:06 Initializing connections...
2015/02/04 09:51:06 Initializing connection arduino ...
2015/02/04 09:51:06 Initializing devices...
2015/02/04 09:51:06 Initializing device led ...
2015/02/04 09:51:06 Starting Robot bot ...
2015/02/04 09:51:06 Starting connections...
2015/02/04 09:51:06 Starting connection arduino on port /dev/ttyACM0...
2015/02/04 09:51:12 Starting devices...
2015/02/04 09:51:12 Starting device led on pin 13...
2015/02/04 09:51:12 Starting work...
^C2015/02/04 09:51:35 Stopping Robot bot ...
dkinzer09:51[~/go/src/github.com/hybridgroup/gobot/examples]
(examples)(master)$./firmata_blink 
2015/02/04 09:51:47 Initializing Robot bot ...
2015/02/04 09:51:47 Initializing connections...
2015/02/04 09:51:47 Initializing connection arduino ...
2015/02/04 09:51:47 Initializing devices...
2015/02/04 09:51:47 Initializing device led ...
2015/02/04 09:51:47 Starting Robot bot ...
2015/02/04 09:51:47 Starting connections...
2015/02/04 09:51:47 Starting connection arduino on port /dev/ttyACM0...
```

saljam, 2015-03-31:
I can reproduce this but only on Linux. OS X seems to work fine.

Also worth mentioning that unbinding and rebinding the USB serial driver doesn't seem to affect it. Only unplugging and plugging back in again or pressing the reset button on the Arduino seems to do the job.

dkinzer, 2015-03-31:
Yes, I should have mentioned that I was using Ubuntu 14.04 LTS when I ran into this issue.

saljam, 2015-03-31:
OK it seems that when the Firmata firmware is reset it preemptively sends over its firmwareQuery answer, which triggers the rest of the initialisation process in the package.

But for some reason, the reset command doesn't actually register on my Arduino when send from a Linux host :-/ Explicitly sending a firmwareQuery when connecting works around this, however. I'll send a PR in a bit.

zankich, 2015-04-06:
@dkinzer does https://github.com/hybridgroup/gobot/pull/180 fix the problem you're experiencing?

dkinzer, 2015-04-07:
Sorry, I can't check this until tomorrow.

dkinzer, 2015-04-08:
@saljam @zankich it works, but not consistently.  Almost invariable on the first attempt it fails.  I control C out of that and try again and then it works.  Sometimes have to even try several failed attempts before it works.

By the way, what's the canonical way of trying out patches in go projects.  For this one I just manually made the change on my local firmata.go file.

saljam, 2015-04-08:
@dkinzer thanks for testing! I'll try to make it produce the same behaviour as that when I get some time. When I tried it it seemed to work fine here.

To try out a pull request you could run:

    $ git fetch origin pull/<pull-request-id>/head:<new-branch-name>
    $ git checkout <new-branch-name>

and rebuild the project. The [GitHub documentation](https://help.github.com/articles/checking-out-pull-requests-locally/) has more about this.

dkinzer, 2015-04-08:
@saljam thanks.  The problem regarding testing of a patch had more to do with the Go System then the technicality of pulling in new code and running it, because go imports take repo source direclty (i.e "github.com/hybridgroup/gobot/platforms/firmata").

saljam, 2015-04-08:
The go tool checks out the repository into $GOPATH if it doesn't already exist, and that's pretty much all it knows about version control. Other than that, the import path is just the path to the package source files. So if you `cd $GOPATH/github.com/hybridgroup/gobot` then fetch and checkout the pull request in there it will use the new version.

dkinzer, 2015-04-08:
cool. thanks!

zankich, 2015-06-30:
@dkinzer the firmata package has been updated, can you please pull the latest master branch and test, thanks!

saljam, 2015-07-01:
FWIW the latest master works for me. Thanks again.

edgarsilva, 2015-07-04:
Closing this since it seems fixed in master and latest release.

closed 2015-07-04 by edgarsilva