hybridgroup/gobot #174: beaglebone usr led path incorrect

blalor, 2015-02-04:
Trying out gobot with a white Beaglebone running Debian.  The path to the usr leds on my device is `/sys/devices/ocp.3/gpio-leds.7/leds/beaglebone:green:`, but the Beaglebone adaptor is hard-coded to `/sys/devices/ocp.3/gpio-leds.8/leds/beaglebone:green:`.  So the blinking doesn't happen (and worse, there's no error).

I haven't found a canonical source for this info, but from [this](http://derekmolloy.ie/beaglebone-controlling-the-on-board-leds-using-c/) it appears (and I've verified) that there are symlinks in `/sys/class/leds` that point to the appropriate `gpio-leds.X` location.  Also referenced [here](http://robotic-controls.com/book/export/html/69).

If there's not already a fix in the works (haven't seen any other issues filed for this), I'll whip up a PR.

zankich, 2015-02-04:
@blalor it seems like doing a `glob` similarly to how the `ocp` and `slots` are figured out should be an easy fix. I would gladly accept a pr to fix this issue :+1:

deadprogram, 2015-11-07:
Hi, @blalor 

Did you ever create a fix for this issue, and if so, can you submit a PR?

Thanks!

deadprogram, 2016-11-27:
This issue should now be addressed in the https://github.com/hybridgroup/gobot/tree/feature/beaglebone-4 branch.

deadprogram, 2016-11-30:
Here is a PR for any comments: https://github.com/hybridgroup/gobot/pull/342

deadprogram, 2016-12-03:
This has been merged into the `dev` branch. Thanks everyone!

closed 2016-12-03 by deadprogram