hybridgroup/gobot #176: problem in blink led gobot

myalexer, 2015-02-21:
hi
my platform is arduino - os : win7 
my code :
package main

import (
	"time"
	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/platforms/firmata"
	"github.com/hybridgroup/gobot/platforms/gpio"
)

func main() {
	gbot := gobot.NewGobot()

	firmataAdaptor := firmata.NewFirmataAdaptor("arduino", "COM15")
	led := gpio.NewLedDriver(firmataAdaptor, "led", "13")
	
	work := func() {
		gobot.Every(1 * time.Second, func() {
			led.Toggle()
		})
	}
	*/
	robot := gobot.NewRobot("bot3",
		[]gobot.Connection{firmataAdaptor},
		[]gobot.Device{led},
		work,
	)
	
	gbot.AddRobot(robot)

	gbot.Start()
}

all things is ok 
i type go run blink.go 
it return :
2015/02/21 00:07:40 Initializing Robot bot3 ...
2015/02/21 00:07:40 Initializing connections...
2015/02/21 00:07:40 Initializing connection arduino ...
2015/02/21 00:07:40 Initializing devices...
2015/02/21 00:07:40 Initializing device led ...
2015/02/21 00:07:40 Starting Robot bot3 ...
2015/02/21 00:07:40 Starting connections...
2015/02/21 00:07:40 Starting connection arduino on port COM15...

program start successful but led do not Toggle every 1 second
how do i do?

zankich, 2015-02-23:
@myalexer Is your arduino flashed with firmata and are you sure that COM15 is the correct port number? I just tested on windows 7 and it worked for me...

myalexer, 2015-02-24:
yes i test it with arduino program - in arduino ide port is COM15  - blink program with arduino syntax is ok and it work

myalexer, 2015-02-24:
how can i found COM port number?

mukosolu, 2015-04-09:
Hi, I also struggled with this problem. it might be an idea for this page: http://gobot.io/documentation/getting-started/ to also tell users to run the firmata example first.

myalexer, 2015-04-12:
i test it but it can't work
i found port . but led not toggle every second 
i think it's bug

LaraChicharo, 2015-04-27:
I have the same problem on Arduino uno, but Ubuntu 14.10. It blinks three times no matter what port I set, even when led is not in that port. According to me, my serial port is correct 'ttyACM0'. If I get something I will let you know.

LaraChicharo, 2015-04-27:
I can run it now!
My problem was that I was not flashed the Arduino with firmata.
http://firmata.org/wiki/Main_Page#Arduino_Uno

I'm not really sure, but, in gobot's getting started documentation there is nothing about firmata.

zankich, 2015-04-27:
@LaraChicharo I'll add more documentation about steps for loading firmata on your arduino, sorry for the confusion!

myalexer, 2015-05-02:
zankich please help me for solved this problem 
i start 2 month ago , but still i have problem in blink led !!!!

pedromorgan, 2015-06-20:
Yea I;m kinda stuck with Firmata not working also,... amd confused..

edgarsilva, 2015-07-03:
Hi guys,

First things first, the process to get the blink example working is as follows:

### Upload Firmata Sketch to the Arduino.
You can do this using the Arduino IDE or using (gort.io)[gort.io].

When using the Arduino IDE just click on file > examples > firmata > StandardFirmata and upload that sketch to the arduino.

When using (gort.io)[gort.io] first install it, then run:
```
# This installs avrdude, which allows to upload the firmata sketch to the arduino
gort arduino install

# Scan serial port for arduino serialport address
gort scan serial

# Upload firmata protocol/sketch to the correct port the arduino is connected to
$ gort arduino upload firmata /dev/ttyACM0
```

### Compile and run arduino_blink.go

Make sure you are using the correct port, in the case of windows you can check the port is the correct one if you can upload an example (like blink sketch) in the arduino IDE. If it uploads and works you should be good. Just update the port in the gobot example and compile.

In Linux is even easier to identify the serial port by using either `gort scan serial` or just checking `/dev/ttyACM*`.

Compile and run the program, after that it should work fine.

Let me know if that helps.

deadprogram, 2015-11-07:
Hi, everyone

I think the docs located here are sufficient: https://github.com/hybridgroup/gobot/blob/dev/platforms/firmata/README.md

If not, someone can re-open this issue. Thanks, and closing.

closed 2015-11-07 by deadprogram

kedare, 2017-01-07:
Same problem here, Arduino Uno with the firmata firmware flashed fine ( I could use pyFirmata to test), nothing happens, I added some logs in the program to have more information :

```
package main

import (
	"fmt"
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/firmata"
)

func main() {
	firmataAdaptor := firmata.NewAdaptor("/dev/cu.usbmodem1411")
	led := gpio.NewLedDriver(firmataAdaptor, "13")

	work := func() {
		gobot.Every(1*time.Second, func() {
			fmt.Println("Updating LED")
			led.Toggle()
		})
	}

	robot := gobot.NewRobot("bot",
		[]gobot.Connection{firmataAdaptor},
		[]gobot.Device{led},
		work,
	)

	fmt.Println("Starting")
	robot.Start()
	fmt.Println("Started")
}
```

```
2017/01/07 22:43:22 Initializing connections...
2017/01/07 22:43:22 Initializing connection Firmata ...
2017/01/07 22:43:22 Initializing devices...
2017/01/07 22:43:22 Initializing device LED ...
2017/01/07 22:43:22 Robot bot initialized.
Starting
2017/01/07 22:43:22 Starting Robot bot ...
2017/01/07 22:43:22 Starting connections...
2017/01/07 22:43:22 Starting connection Firmata on port /dev/cu.usbmodem1411...
```

It's not even running the work function.

The TTY port is correct (Detected by Arduino IDE and working fine with pyFirmata)

What could be the cause ?

kedare, 2017-01-07:
Ok I found the issue, looks like for master part is missing from the example code, here is the working one :

```
package main

import (
	"fmt"
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/firmata"
)

func main() {
	gbot := gobot.NewMaster()
	firmataAdaptor := firmata.NewAdaptor("/dev/cu.usbmodem1411")
	led := gpio.NewLedDriver(firmataAdaptor, "13")

	work := func() {
		gobot.Every(1*time.Second, func() {
			fmt.Println("Updating LED")
			led.Toggle()
		})
	}

	robot := gobot.NewRobot("bot",
		[]gobot.Connection{firmataAdaptor},
		[]gobot.Device{led},
		work,
	)

	gbot.AddRobot(robot)

	fmt.Println("Starting")
	gbot.Start()
	fmt.Println("Started")
}
```

deadprogram, 2017-01-08:
Hi @kedare the original example code works for me, albeit by changing my serial port to `/dev/ttyACM0` as I am on Linux.

```
$ go run ./examples/check.go 
2017/01/08 17:29:05 Initializing connections...
2017/01/08 17:29:05 Initializing connection Firmata ...
2017/01/08 17:29:05 Initializing devices...
2017/01/08 17:29:05 Initializing device LED ...
2017/01/08 17:29:05 Robot bot initialized.
Starting
2017/01/08 17:29:05 Starting Robot bot ...
2017/01/08 17:29:05 Starting connections...
2017/01/08 17:29:05 Starting connection Firmata on port /dev/ttyACM0...
2017/01/08 17:29:10 Starting devices...
2017/01/08 17:29:10 Starting device LED on pin 13...
2017/01/08 17:29:10 Starting work...
Updating LED
Updating LED
Updating LED
Updating LED
Updating LED
Updating LED
```

Since the v1.0 release you can use "Classic Gobot" as the first example does, so no need for the `Master()` part.

I then tested the code on OSX, and both examples worked with my serialport `/dev/tty.usbmodem1421`. I suggest you use the `tty` vs the `cu` version of the serial port on OSX.

Hope that helps!