hybridgroup/gobot #181: Parallella support?

samuell, 2015-04-01:
I'm highly intrigued by both the gobot framework and the [parallella platform](https://www.parallella.org/), and just wanted to ask if there are any plans to support it in the near future?

deadprogram, 2015-04-01:
That looks really cool. Seems like it would be relatively easy to add support for it via standard Linux GPIO and just getting the mappings setup in similar fashion to how Beaglebone Black is https://github.com/hybridgroup/gobot/blob/master/platforms/beaglebone/beaglebone_adaptor.go#L37-L125

deadprogram, 2016-04-18:
Paging Mr. @rayhightower here! :)

befedo, 2017-08-21:
Is there any work-in-progress here?!

deadprogram, 2017-08-22:
Hi @befedo I do not have this particular hardware myself for testing, but I'd be happy to help someone who does.

closed 2025-03-21 by gen2thomas