hybridgroup/gobot #182: Docs clarification: what is the MCP?

Symmetric, 2015-04-15:
I've had a look at the docs, and this project looks very useful -- but the demo examples don't discuss the high-level architecture very thoroughly.

There's mention of an MCP in a few places, but this isn't really covered in the docs. After a bit of digging and watching your videos and so on, I think I've got a grasp on these concepts -- but I think it would be useful to extend the docs to clearly cover a few of these points:

* What is the MCP?
 * For a "Hello world" usecase, probably a laptop
 * For an IOT household sensor project, I suppose this could be a Raspberry Pi?
 * Any other use cases you have envisioned?
* Can you load a program onto a microcontroller and then run it in standalone mode?
 * The docs seem to suggest that I need to keep my laptop (MCP?) connected all the time, rather than the Arduino way of compiling and loading a program onto the microcontroller. This is a big difference which isn't explicitly called out, and I think should be clarified.
 * I think a corrolary of this is that if I want to use an Arduino with GoBot, it must be connected to another device at all times.
 * What about ARM microcontrollers (such as the Teensy) which can run Go code natively? It looks like GoBot doesn't currently support running these guys in standalone mode.

If any of these points are wrong/incomplete please correct me, this is just what I've gleaned from a bunch of digging on your site.

zankich, 2015-04-15:
@Symmetric so basically the `MCP` is what controls and coordinates all of your [robots](http://godoc.org/github.com/hybridgroup/gobot#Robot). It is represented as the [gobot](http://godoc.org/github.com/hybridgroup/gobot#Gobot) type. When you do a `NewGobot()` you are instantiating a new `MCP` which you can then add robots to. The `MCP` can run locally on a computer, in the case of a laptop or raspberry pi, or it can run in a cloud instance, and remotely control robots, or communicate with other `MCP` instances to do some work. The `MCP` is the part that allows the [api](http://godoc.org/github.com/hybridgroup/gobot/api) to access your robots, and does the auto initialization and cleanup of your robots in your program. This architecture [diagram](http://cylonjs.com/images/elements/architecture-mcp.png) might help understand what's going on. Our sister project cylon.js has a section on the website discussing the architecture, which needs to be added to the gobot site as well. You can check it out [here](http://cylonjs.com/documentation/architecture/) and even though it's talking about cylon, the general architecture is still the same.

As far as running gobot on a microcontroller, that is currently not possible due the limitations in the go language. Go requires an operating system to run, so if your board does not have an OS then you can't run go on it. This however does not mean that you can't use gobot to control a microcontroller. If the microcontroller has a remote interface, such as firmata, littlewire, etc, then you can use gobot to remotely control the device. In the case of a teensy or an arduino, if you add something like a zigbee or wifi radio, you can use gobot to remotely control them from a central server, which will hold your app logic and coordinate your devices.

I hope this helps answer some of your questions. The architecture diagram and info really needs to be added to the website so that more of this stuff is clear.

deadprogram, 2016-10-21:
The `dev` branch of Gobot has many changes, one of which is that you can now run Gobot just using Robots aka "Classic Gobot". You can also use "Master Gobot" if what you want is to control multiple robots or use the API.

I am going to close this issue, please reopen if needed. Thanks.

closed 2016-10-21 by deadprogram