hybridgroup/gobot #190: Is Orbotix Ollie supported ?

coxx, 2015-06-26:
Hello guys!
Can you explain is Ollie supported or just Sphero ?

deadprogram, 2015-06-26:
Hi, @coxx 

We do not have Ollie support yet, due to still adding core Bluetooth LE support. The Sphero is Bluetooth classic, and so was easier to add support.

Hope that helps!

ryanfaerman, 2016-03-03:
Any updates on Ollie/Bluetooth LE support?

deadprogram, 2016-03-03:
Hi, @ryanfaerman I am working on BLE support, hope to have something pretty soon.

Then the Ollie/BB-8 API will be next!

ryanfaerman, 2016-03-03:
Is there anything I can do to help?

I've recently purchased a BB8 sphero and I'm debating returning it and getting a Sprk so that I can play with things.

deadprogram, 2016-03-03:
Checkout my work in progress branch here: https://github.com/hybridgroup/gobot/tree/feature/ble-wip

deadprogram, 2016-10-21:
Hi, everyone

Initial support for Ollie & BB-8 is now in Gobot 0.12+ sorry I did not update this issue. Please check it out!

closed 2016-10-21 by deadprogram