hybridgroup/gobot #200: Why return slice of errors?

saulshanabrook, 2015-07-10:
I noticed that in lots of the code you return `[]error` instead of just `error`.

I have never seen this pattern before. It is a bit more troublesome to handle and return. In places where it is necessary to return multiple, maybe use [this idea](https://groups.google.com/forum/#!msg/golang-nuts/zE-CrCoeziw/jJCOOt4Ob_kJ)?

> When you can have multiple errors, you can do something like http://play.golang.org/p/X2Q2aVJweW (a []error that fulfills the error interface itself).

mattetti, 2015-07-12:
I would recommend https://github.com/hashicorp/go-multierror I know this adds an extra dependency but this package is used in packer and many other popular great packages.

joushou, 2015-07-21:
Most of the places where I saw []error, it did not seem to serve a purpose, suggesting that one should simply return one error, and in cases with multiple errors, fail on first error.

Let's wait to hear if the author had a good reason we're missing for this slightly peculiar construct.

zankich, 2015-07-21:
The places where Gobot returns a `[]error` is on the `Connect` and `Finalize` methods. The reason for this is to allow partial failures in connection or disconnection routine, but still continue to operate, instead of bombing out. An example of this in a `Finalize` is in the `Edison` [adaptor](https://github.com/hybridgroup/gobot/blob/d5605838ddb8ee91ec81ab00386c965a5d4c0737/platforms/intel-iot/edison/edison_adaptor.go#L276-L301). If one part of the cleanup process fails, we don't want to completely stop, but clean up whatever we can and deal with the errors as they arise. I would like to return a custom error in those cases instead of a `[]slice`, but when I was originally refactoring the code, the error system was really bad and this was the first step in properly handling errors throughout the packages.

deadprogram, 2016-11-07:
Thanks everyone for the good discussion. Took a while, but I've finally created https://github.com/hybridgroup/gobot/pull/336 which just returns a normal error instead of a slice of errors, and uses multierror as suggested by @mattetti in the case where a list of errors is really needed.

Please feel free to add any comments to the PR, otherwise it will be merged into the dev branch later this week.

deadprogram, 2016-11-13:
This has now been merged into the `dev` branch. Thanks everyone!

closed 2016-11-13 by deadprogram