hybridgroup/gobot #205: PWM exported functions for Intel Edison

mattetti, 2015-07-11:
To be able to properly use the PWM possibilities of the Edison, one needs to set the `period` and `duty cycle` to create a pulse at a given frequency:

```
If you want a 1.6kHz wave that is a 625 microsecond period or 626000 nanoseconds (assuming a 50% duty cycle).
 
echo -n "625000" > /sys/class/pwm/pwmchip0/pwm3/period
echo -n "312500" > /sys/class/pwm/pwmchip0/pwm3/duty_cycle
```

Unfortunately, right now, gobot developers don't have access to period and duty cycle, as a matter of fact, they have almost no access to anything pwm related and only `func (e *EdisonAdaptor) PwmWrite` seems exposed (and it does a lot of magic).

My suggestion is to export `pwmPin` and its methods so developers interested in going a bit lower level can have fun ;)

mattetti, 2015-07-11:
@zankich @deadprogram This is currently a blocker for me to generate a proper audio signal via PWM. 
For anyone else looking at this issue, here is a good document on PWM: https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM

mattetti, 2015-07-13:
Adrian and I discussed this issue in RL and he suggested to create a new PWM interface for devices offering access to period and duty cycles

deadprogram, 2017-05-03:
Finally addressed this is part of the working done in reference to https://github.com/hybridgroup/gobot/pull/410

There is now a PWMPin that can used with all of the Linux SoC boards for more granular control over the PWM on each board.

For example:

```
a := edison.NewAdaptor()
pin := a.PWMPin("3")

// get period
period := pin.Period()

// set to 50% duty cycle
duty := period / 2
pin.SetDutyCycle(duty)
```

This is all available in the `dev` branch. I will close this issue as soon as it lands in `master` as part of the next release.

deadprogram, 2017-05-10:
This code has now been released as part of Gobot 1.5 so please either open a new issue if needed with any specifics, or reopen this one.

Thanks.

closed 2017-05-10 by deadprogram