hybridgroup/gobot #210: LIDARLiteDriver returning nonsense values

mwhooker, 2015-07-12:
Hi All,

I was working on this during the hack day. I was able to do a little more debugging, so thought I'd open an issue and give some more info.

using latest from master, `LIDARLiteDriver.Distance` makes a write to 0x8f, as per the [docs](http://kb.pulsedlight3d.com/support/solutions/articles/5000548624-quick-start-guide).

> Read:  Register 0x0f, returns the upper 8 bits of distance in cm, register 0x10, returns the lower 8 bits of distance in cm.  (Optionally a 2-Byte read starting at 0x8f can be done)

When it goes to make a read, it gets back nonsense values, such as `[0x00, 0x00]` or `[0x01, 0x01]`.

After some poking, it looks like it's grabbing the upper 8 bits of the distance twice. I can make a read to 0x10 and 0x0f separately, and get back the expected values.

I'll do that for now (I can post code if anyone's interested), but please lmk if I can help debug more.

zankich, 2015-07-12:
@mwhooker Thanks for reporting the issue! I pushed an [update](https://github.com/hybridgroup/gobot/commit/c3bc23521878747e3e1ba13059abb78142ae18f7) to the driver and everything should be good now

mwhooker, 2015-07-12:
awesome! looks like we ended up doing the same thing https://github.com/mwhooker/speedo/blob/b6a116c9403cd26db63183fe1aaea9f006d43dc7/lidar.go

closed 2015-07-16 by zankich