hybridgroup/gobot #211: errors when reading from LIDARLiteDriver and JHD1313M1Driver

mwhooker, 2015-07-12:
code to reproduce is here https://github.com/mwhooker/speedo/tree/b6a116c9403cd26db63183fe1aaea9f006d43dc7

Every time I run the program I get an error, although the behavior varies:

```
root@edison:~# ./speedo
2015/07/12 07:04:36 Initializing Robot screenBot ...
2015/07/12 07:04:36 Initializing connections...
2015/07/12 07:04:36 Initializing connection edison ...
2015/07/12 07:04:36 Initializing devices...
2015/07/12 07:04:36 Initializing device lidar ...
2015/07/12 07:04:36 Initializing device screen ...
2015/07/12 07:04:36 Starting Robot screenBot ...
2015/07/12 07:04:36 Starting connections...
2015/07/12 07:04:36 Starting connection edison...
2015/07/12 07:04:36 Starting devices...
2015/07/12 07:04:36 Starting device lidar...
2015/07/12 07:04:36 Starting device screen...
[  505.390547] i2c-designware-pci 0000:00:09.1: i2c_dw_handle_tx_abort: lost arbitration
2015/07/12 07:04:36 Error: Robot "screenBot": Device "screen": write /dev/i2c-6: resource temporarily unavailable
2015/07/12 07:04:36 Stopping Robot screenBot ...
root@edison:~# ./speedo
2015/07/12 07:04:40 Initializing Robot screenBot ...
2015/07/12 07:04:40 Initializing connections...
2015/07/12 07:04:40 Initializing connection edison ...
2015/07/12 07:04:40 Initializing devices...
2015/07/12 07:04:40 Initializing device lidar ...
2015/07/12 07:04:40 Initializing device screen ...
2015/07/12 07:04:40 Starting Robot screenBot ...
2015/07/12 07:04:40 Starting connections...
2015/07/12 07:04:40 Starting connection edison...
2015/07/12 07:04:40 Starting devices...
2015/07/12 07:04:40 Starting device lidar...
2015/07/12 07:04:40 Starting device screen...
2015/07/12 07:04:40 Starting work...
Distance (in) 0
Distance (in) 0
Distance (in) 0
^C2015/07/12 07:04:42 Stopping Robot screenBot ...
```

zankich, 2015-07-12:
@mwhooker the problem is that the i2c rgb led on the jhd1313m1 has the same i2c address of the lidar....

mwhooker, 2015-07-12:
that's unfortunate. Is there any chance of a workaround? perhaps I can take a screwdriver to the lcd i2c controller...

zankich, 2015-07-12:
@mwhooker there is a [pwm](https://github.com/PulsedLight3D/LIDARLite_Basics/blob/master/Arduino/LIDARLite_PWM_GetDistance_ContinuousRead/LIDARLite_PWM_GetDistance_ContinuousRead.ino) interface to the lidar, but that will require implementing a `pulseIn` equivalent in Gobot

mwhooker, 2015-07-12:
okay. I think I'll just modify https://github.com/hybridgroup/gobot/blob/f63fab81a3fecf2d9cc6fdfd643625fdb60fd834/platforms/i2c/jhd1313m1_driver.go to not interact with the led, or perhaps a hardware solution. thanks!

mwhooker, 2015-07-13:
above didn't work, which should have been obvious. the i2c slave still listens on the conflicting address

according to http://www.nxp.com/documents/data_sheet/PCA9533.pdf, the slave address is hard coded, but there's another version I can get with a different address (PCA9533/02). I think I'll try to track down one of those. closing, thanks!

I'm not sure this is possible, but it would be neat if gobot could detect i2c collisions and give me an error about it.

closed 2015-07-13 by mwhooker