hybridgroup/gobot #213: Robot.Start should run r.Work() as a goroutine.

pborman, 2015-07-14:
In gobot/robot.go, around line 134, func (r *Robot) Start() calls r.Work().  If r.Work() does not return then Start will hang.  If this hangs then func (r *Robots) Start() will also hang and not call Start on any further Robots.  The solution is to make the call a goroutine.  Patch attached below.

diff --git a/robot.go b/robot.go
index 8775e87..131fade 100644
--- a/robot.go
+++ b/robot.go
@@ -131,7 +131,7 @@ func (r *Robot) Start() (errs []error) {
 	}
 	if r.Work != nil {
 		log.Println("Starting work...")
-		r.Work()
+		go r.Work()
 	}
 	return
 }

pborman, 2015-07-14:
The patch, without being marked down, is:

```
diff --git a/robot.go b/robot.go
index 8775e87..131fade 100644
--- a/robot.go
+++ b/robot.go
@@ -131,7 +131,7 @@ func (r *Robot) Start() (errs []error) {
 	}
 	if r.Work != nil {
 		log.Println("Starting work...")
-		r.Work()
+		go r.Work()
 	}
 	return
 }
```

zankich, 2015-07-14:
@pborman can you please create a pull request with this patch?

joushou, 2015-07-21:
The suggestion in #216 (split gobot.Start into gobot.Start and gobot.Stop) would most likely deprecate Work (it would no longer serve a purpose), which I find to be a bit of a peculiar construct.

zankich, 2015-07-21:
@joushou the work routine is a primitive of a `Robot` once it has successfully `Started`. When you start your `Robot` it has it's `Work` that it does. Adding the `Gobot.Start/Stop` wouldn't change that behavior.

joushou, 2015-07-21:
Ah, I think I was a bit too quick and rough in my wording. Work is called is called at the end of Robot.Start. It is currently necessary because Gobot.Start, which has the responsibility of calling Robot.Start, blocks, and doesn't terminate before os.Interrupt is issued.

Now, if Gobot.Start only called Robot.Start, or a Robot.Stop is made available so that Gobot.Start wasn't necessar, then normal program flow can follow immediately after Robot.Start/Gobot.Start, effectively making Work a slightly funky construct. It would make:

       r := gbot.NewRobot(lalala, lalala, func(){ fmt.Println("DONE!") })
       r.Start()

completely equivalent to:

       r := gbot.NewRobot(lalala, lalala)
       r.Start()
       fmt.Println("DONE!")

Therefore, while the issue of changing Gobot.Start/Stop doesn't directly affect the behaviour of Work, it does make it somewhat redundant, and I was suggesting that its presence should maybe be reconsidered.

Does my message make more sense this way?

deadprogram, 2015-10-31:
@joushou the `Work` is the user's defined function for whatever the `Robot` is supposed to do. `robot.Start` should mean "robot, start doing your defined work".

As such, "work" is an important part of the semantics of the design patterns behind Gobot.

Anyhow, this is somewhat off topic of the original issue. @pborman any chance of getting a PR against `dev` branch?

Thanks everyone!

deadprogram, 2016-10-21:
There has been substantial changes to how Gobot handles Work. Cross-posted from https://github.com/hybridgroup/gobot/issues/283#issuecomment-255311798

There has been subtancial changes to how Gobot creates and manages the goroutines used. Please take a look at the dev branch to see what we've been doing: https://github.com/hybridgroup/gobot/tree/dev

In short:

    every Robot's Work is now run inside of a separate goroutine that exists for the life of the Robot.
    every Device's On and Once event handlers are each run in a separate goroutine that exists for the life of the Device.
    channels are now used for Events, and the implementations of On and Once use the event channels as their implementation to communicate when things happen.

The dev branch has a lot of breaking changes, but I think you will find the syntax is simpler, and the implementation much cleaner and more concurrent by default.

I am going to close this issue, please reopen if so needed. Thanks everyone!

closed 2016-10-21 by deadprogram