hybridgroup/gobot #216: gobot should not intercept os.Interrupt

joushou, 2015-07-20:
gobot should not attempt to "own" the main application, by making blocking code that starts messing with application-wide signal handlers. A more Go-style external API would be:

        gobot.Start()
        defer gobot.Stop() // This would then call the connection finalizers

This would let users decide on their own shutdown mechanisms, rather than have gobot try to own the main application.

My personal idea would be to actually just expose and use Start and Stop on each robot, which would effectively make gobot.Start and gobot.Stop mere convenience wrappers that just iterated them (which could either be killed completely or left behind for convenience). Does that idea sound bad?

zankich, 2015-07-20:
@joushou I completely agree and refactoring that section of code has been on my radar for a while.

joushou, 2015-07-21:
I'm glad! Would a PR on the simple part of this (split gobot.Start into Start and Stop) be of interest, or would you rather wait for the large refactoring to avoid breaking API too often?

zankich, 2015-07-21:
@joushou that is what the `dev` branch is for! There are actually quite a few breaking changes that I want to make.  

This change really requires edits in both the `Gobot` and `Robot` structs. 

This [code](https://github.com/hybridgroup/gobot/blob/master/gobot.go#L75-L87) should pretty much be turned into a `Robot.Stop()` method, similarly to the [Start](https://github.com/hybridgroup/gobot/blob/master/robot.go#L121)

And the `Gobot.Stop()` method should iterate over each `Robot` and call it's stop.

deadprogram, 2015-10-31:
Hi @joushou I put back part of the original code for handling the interrupt, in order to keep the default behaviour and not force all users to add this code to even a simple Gobot program in this commit https://github.com/hybridgroup/gobot/commit/632058c47aea3be68bfaa8b4dd3297d0959960c5

The `AutoStop == true` is the default, but you can set it to false in the `gobot.NewGobot` constructor to write your own exit handling code.

I think this seems to accommodate the use case you want.

deadprogram, 2016-04-18:
This was accommodated in a release last year. Closing, please reopen if needed.

closed 2016-04-18 by deadprogram