hybridgroup/gobot #217: Use time.Sleep where applicable

joushou, 2015-07-21:
A lot of places seem to "abuse" the time.After() timeout channel, which is most useful when used as a timeout on a select. When one does not care about channels, time.Sleep() should save you some locks and be clearer.

(Sorry if I'm slowly spamming your issues, but if I don't write it down, I forget it before I get around to making a PR!)

deadprogram, 2016-11-04:
Hi, @joushou there have been some recent related changes in this area as part of https://github.com/hybridgroup/gobot/pull/332 what do you think?

joushou, 2016-11-04:
There are still a few `<-time.After()` calls left (a lot in examples, but a few in "real" code too - a quick `git grep` shows a good list), where a `time.Sleep()` was intended, but it certainly has improved.

deadprogram, 2016-11-05:
OK, I think this commit should now entirely take care of the `time.Sleep()` thing: https://github.com/hybridgroup/gobot/commit/9faf9eb52a5fb57bb0ff654b69ef5d9d04d356b3 

Thanks for the feedback, @joushou

deadprogram, 2016-11-05:
I'm going to close this now, please reopen if needed.

closed 2016-11-05 by deadprogram