hybridgroup/gobot #219: Aggressive spam with firmata

joushou, 2015-07-21:
I haven't debugged this issue deeper, but making a simple robot with firmata, and making an analog reader with interval of 5 seconds causes the arduino to send aggressive rates of messages, which does not seem right (if I log in firmata/client.go@process, my stdout is spammed to death, and the TX light is practically permanently lit on the arduino).

Unsure if bug in firmware or gobot, or if but at all.

deadprogram, 2015-08-25:
Hi, @joushou 

Using Firmata requires polling the serial port that is connected to the Arduino. THis is needed to determine if there are any pin changes, new i2c data, etc. As such, a fairly constant rate of communication is required with the serial port being used when using the Firmata protocol.

Please let me know if I can provide any further info, and reopen this issue if so. Thanks!

joushou, 2015-08-25:
If I read the implementation correctly, you're not supposed to poll any of that. REPORT_DIGITAL is supposed to handle info about digital pins, I2C will report back if it's set to continous mode, etc., etc.. It's meant to be a push protocol, not polling.

Of course, I may be wrong, but that's what it looks like from the StandardFirmata Arduino implementation.

joushou, 2015-08-25:
Also, I can't reopen issues. :/

deadprogram, 2016-11-04:
Hi @joushou the implementation of the Firmata's digital and analog readers should have been awaiting report notifications for a while now. I did not testing myself recently and did not see the behavior you are describing. This time I really am going to close, please open a new issue if you see this behavior. Thanks again!

closed 2016-11-04 by deadprogram

joushou, 2016-11-04:
Sounds good to me. I'll reopen if I get around to testing and trying this again.