hybridgroup/gobot #222: Cannot run non-main package

4ur3l13n, 2015-08-24:
trying to (re) install gobot on beagleblone (archlinux) because of the issue https://github.com/hybridgroup/gobot/issues/221 I get a new issue.
bot@icecube:~/go/src/github.com/hybridgroup/gobot/platforms/beaglebone$ ls
beaglebone_adaptor.go  beaglebone_adaptor_test.go  doc.go  LICENSE  pwm_pin.go  README.md
bot@icecube:~/go/src/github.com/hybridgroup/gobot/platforms/beaglebone$ go build beaglebone_adaptor.go
# command-line-arguments
./beaglebone_adaptor.go:130: undefined: pwmPin
bot@icecube:~/go/src/github.com/hybridgroup/gobot/platforms/beaglebone$ go run beaglebone_adaptor.go 
go run: cannot run non-main package
bot@icecube:~/go/src/github.com/hybridgroup/gobot/platforms/beaglebone$ uname -a && go version
Linux icecube 4.1.6-1-ARCH #1 Thu Aug 20 22:52:03 MDT 2015 armv7l GNU/Linux
go version go1.5 linux/arm
bot@icecube:~/go/src/github.com/hybridgroup/gobot/platforms/beaglebone$

deadprogram, 2015-08-25:
The problem here is that you are trying to run the wrong thing in Gobot. You want to run your own program, that uses the various adaptors/drivers. For example, you should be trying to run one off the BBB examples.

Hope that helps!

4ur3l13n, 2015-08-28:
Thanks for your light!
Sorry for the disturb.
Sure that helps!

closed 2015-08-28 by 4ur3l13n