hybridgroup/gobot #224: how to direct upload to arduino

alioygur, 2015-08-27:
hi,

how can i direct upload my go project to arduino ? I tried that,

```
gort arduino blink gobot /dev/ttyACM0

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "gobot"
avrdude: invalid record at line 5316 of "gobot"
avrdude: read from file 'gobot' failed

avrdude done.  Thank you.

2015/08/27 20:27:03 exit status 1
```

deadprogram, 2015-08-27:
Hi, @alioygur 

You cannot run Golang directly on an Arduino. Gobot communicates with the Firmata program running on an Arduino microcontroller, by using a serial interface.

Please let me know if that is clear. Thank you!

alioygur, 2015-08-27:
So it doesn't work standalone ?

stewart, 2015-08-27:
Hey @alioygur - Arduinos are [microcontrollers](https://en.wikipedia.org/wiki/Microcontroller), and incapable of hosting an operating system or running Go code directly. To provide support in Gobot, we leverage the [Firmata protocol](https://www.arduino.cc/en/Reference/Firmata) to communicate with a sketch the Arduino _is_ capable of running. We then send commands to Firmata over a serial port the Arduino is connected to, and Firmata executes them for us.

However, on more complex single-board Linux computers (Beaglebone Black, Intel Edison, Raspberry Pi, etc), Gobot can be installed and run directly on-board.

alioygur, 2015-08-27:
Okey, thank you so much

alioygur, 2015-08-27:
Dear, can I communicate with it over network instead of USB?

deadprogram, 2015-08-28:
People have done this several different ways, usually with some cloud-based component such as how Particle does it. I do not have any specific links offhand, I will look into it.

deadprogram, 2017-01-04:
Hi @alioygur please checkout the recent support added to the `dev` branch for the ESP8266 using `firmata.TCPAdaptor()` here: https://github.com/hybridgroup/gobot/blob/dev/examples/wifi_firmata_blink.go

This will also work with any Arduino that is using a WiFi shield and has the Wifi variation of ConfigurableFirmata loaded onto the board.

alioygur, 2017-01-04:
good job!

deadprogram, 2017-01-09:
Experimental support for this was just released as part of v1.1.0. Now closing.

closed 2017-01-09 by deadprogram