hybridgroup/gobot #227: support camera

brunoksato, 2015-09-04:
Gotbot support camera work with arduino

example drive camera OV9655 

have support ? thanks

deadprogram, 2015-09-05:
Hi, @brunoksato we currently use the "Firmata" protocol with the Arduino platform. You would need to either extend Firmata, or use some different custom firmware on the Arduino, to support the digital interface used by most of these cameras (SCCB).

We do have OpenCV support, but that is running on a Linux system, not on the Arduino itself. Perhaps that might suit your needs?

brunoksato, 2015-09-06:
i'm need communication for camera + motor + arduino, need to make the communication of these three synchronous way, one depending on the other, using opencv the Gobot I can integrate with Arduino Gobot?

deadprogram, 2015-09-06:
If in that scenario you will have a Linux single-board computer controlling everything, then yes. Since you cannot run Golang on the Arduino itself, that is the only way to do what you describe.

deadprogram, 2015-11-10:
@brunoksato did my last answer address your question?

brunoksato, 2015-11-11:
Sorry delay, yes resolved ! 

Thanks

closed 2015-11-11 by brunoksato