hybridgroup/gobot #230: Contributing when altering the gobot subpackage

trevrosen, 2015-09-27:
Hi folks,
I am wondering how you'd like me to proceed editing things in the "gobot" dummy package. Altering this package means creating PRs that have altered import statements, which would seem to create additional overhead for the team when landing a PR. I was just hoping you could spell out your preferences for these kinds of PRs in CONTRIBUTING.md.

thanks!

-Trevor

deadprogram, 2015-09-27:
Hi @trevrosen thanks for your question, but could you please explain what you mean by 'editing things in the "gobot" dummy package'? I'm sorry I'm not sure what that means.

trevrosen, 2015-09-27:
Gobot contains a package called "gobot" that is imported throughout the library. For instance, the api package imports it. This package is not standalone and appears to be  a method of code organization. These are frequently called "dummy packages" in Go parlance. 

Throughout the Gobot code base, it's imported as a fully qualified path. If you were to fork github.com/hybridgroup/gobot, edits to files in the gobot package won't be picked up in the importing package. 

In other words, I have to rewrite all instances of "github.com/hybridgroup/gobot" to "github.com/trevrosen/gobot" in order to make changes. 

This complicates the PR review/merge process because of the need to be building against the right paths, do a post-merge rename, etc. Different projects handle this in different ways -- I just wanted clarification on how you'd want me to proceed. 

Does that make sense?

((It may be desirable to explore relative paths for solving this issue, which would preserve the package as a code organization mechanism, but not treat it as an external dep as is happening now))

deadprogram, 2015-09-27:
I'm interested if you have any other suggestions on code organization. Relative paths are really frowned on, but I certainly know what you mean. I would probably just symlink or something to avoid that problem?

trevrosen, 2015-09-28:
Yeah if you don't want to do relative paths, the best way is to publish some workflow advice. There's a [pretty good suggestion here](http://blog.campoy.cat/2014/03/github-and-go-forking-pull-requests-and.html), which relies on making your fork into a remote on your copy of the original.

Bottom line is that this is totally solvable w/ some workflow stuff, but there are steps that should probably be taken to make it clear to all contributors what to do. I can take a stab at some changes to CONTRIBUTING.md if you like.

deadprogram, 2015-09-28:
That sounds great, and of course @campoy suggestion nice and simple. Yes, please, PR appreciated!

deadprogram, 2015-09-28:
Oops.

PR #232 (2015-09-29): Add repo setup and PR landing steps
Fixes #230 

Adds information in the contributing guide about how to setup your fork as a remote on your `go get`-created copy of the original (this) repo.

Also adds information helpful to people landing the PR, but this may be superfluous.
commit f50c4052 (2015-09-29): Add repo setup and PR landing steps

Fixes #230
closed 2015-09-29 by deadprogram