hybridgroup/gobot #234: Parrot Rolling Spider support ?

deadjoe, 2015-09-30:
Hi this is amazing project to a golang developer as well as a robot fan.

Just wonder if any plan to support the Rolling Spider from Parrot ? 

thanks,

joe

deadprogram, 2015-10-01:
Hi, @deadjoe we will need BLE support in order to implement support for that platform. We're working on it...

deadjoe, 2015-10-01:
Awesome !

deadjoe, 2016-07-09:
Hi , just want to know if Parrot Rolling Spider supported yet?

deadprogram, 2016-07-09:
Hi @deadjoe we are just adding experimental support in the ble-wip branch. Currently works on OS X but we're trying to get it working on Linux as well.

deadjoe, 2016-07-17:
Thanks , cool !

deadprogram, 2016-07-19:
Since I originally responded to this issue, we released v0.12.x that includes the experimental BLE support and support on OSX. The `dev` branch also includes the Linux support for the minidrone, which will also be released soon.

deadjoe, 2016-08-27:
Thanks, 

I saw the BLE support in the latest release. I am looking for  documents or examples about the BLE support with my rolling spider ? Can't find it from the gobot website and appreciate if you can give a clue.

deadprogram, 2016-12-21:
The docs are now updated with this info. Thanks, everyone.

closed 2016-12-21 by deadprogram