hybridgroup/gobot #236: Constant Values with MPU6050 on Raspberry Pi

ryanskidmore, 2015-10-01:
I've got my MPU6050 connected via I2C to my Raspberry Pi (WHOAMI returns correct address) but I seem to be getting constant values, despite moving the sensor (quite a lot!).

I've attached a stripped down version of my code below in hopes that it's just an issue with my code, rather than deeper in the Gobot library.

If it isn't a code issue on my end, could there potentially be an issue with the way Gobot configures the MPU6050?

```go
MPU6050Work := func() {
		gobot.Every(50*time.Millisecond, func() {
			fmt.Println("Accelerometer", qc.Sensors.MPU6050.MPU6050Driver.Accelerometer)
			fmt.Println("Gyroscope", qc.Sensors.MPU6050.MPU6050Driver.Gyroscope)
			fmt.Println("Temperature", qc.Sensors.MPU6050.MPU6050Driver.Temperature)
			qc.Sensors.MPU6050.Accel.Values = &qc.Sensors.MPU6050.MPU6050Driver.Accelerometer
			qc.Sensors.MPU6050.Gyro.Values = &qc.Sensors.MPU6050.MPU6050Driver.Gyroscope
		})
	}
MPU6050Bot := gobot.NewRobot("MPU6050Bot", []gobot.Connection{qc.RPiAdaptor}, []gobot.Device{qc.Sensors.MPU6050.MPU6050Driver}, MPU6050Work)
qc.GBot.AddRobot(MPU6050Bot)
qc.GBot.Start()

go func(ss *Sensor_MPU6050, dm *map[int64]string) {
		DLInst := &DataLog{
			Accel: ss.Accel.Values,
			Gyro:  ss.Gyro.Values,
		}
		CTime := time.Now().UnixNano()
		DLInstString, err := json.Marshal(structs.Map(DLInst))
		if err != nil {
			color.Red("[!] Failed to marshal DataLog map:" + err.Error())
		} else {
			(*dm)[CTime] = string(DLInstString)
		}
		time.Sleep(25 * time.Millisecond)
	}(qc.Sensors.MPU6050, &DataMap)

```

deadprogram, 2015-10-03:
Hi, @ryanskidmore 

I was just looking at our implementation for MPU6050 in Cylon.js and I see a couple of differences. One in particular being https://github.com/hybridgroup/cylon-i2c/blob/master/lib/mpu6050.js#L113

I will try to find my MPU6050 and play around with this, but if you have a chance to write a little code to send the "wake up" that looks to be worth trying.

ryanskidmore, 2015-10-03:
@deadprogram When I first looked at the driver I initially thought the wakeup instruction was handled by https://github.com/hybridgroup/gobot/blob/dev/platforms/i2c/mpu6050_driver.go#L104, but now that i've looked at it again I can't see any wake up instruction. Just noticed this line though https://github.com/hybridgroup/gobot/blob/dev/platforms/i2c/mpu6050_driver.go#L128 - not entirely sure what this does "setSleepEnabled" gives the impression it's putting it back into sleep mode.

I know that a 0x00 needs to be sent to the PWR_MGMT_1 Register, so I'm guessing that's what's going wrong. I'll try and work on a fix next week sometime.

ryanskidmore, 2015-10-07:
I've had a chance to modify the gobot code but no luck - I removed the set sleep enabled part and send the wake up byte to pwr_mgmt_1 but still getting constant values.

deadprogram, 2015-10-07:
I will play with this when I'm back in the lab later this week. Thanks for reporting.

sgilani, 2015-10-30:
Hi @ryanskidmore 

i think you have the same problem like me before. I think this will help you:

open the file mpu6050_driver.go and change:

const MPU6050_PWR1_SLEEP_BIT = 6 

to 

const MPU6050_PWR1_SLEEP_BIT = 0

commit e5707501 (2016-02-17): Fix #236 & fix #239 by correcting initialization and temperature conversion for MPU-6050
closed 2016-02-17 by deadprogram