hybridgroup/gobot #239: Arduino MPU6050

sgilani, 2015-10-25:
I get only back this when i run the Go Prog. on Windows, i checked the Adress with I2C Scanner fro Adruino and i get back 0x68, some where the communication is broken. 

Any ideas?

Accelerometer {0 0 0}
Gyroscope {0 0 0}
Temperature 0

package main

import (
  "fmt"
  "time"

  "github.com/hybridgroup/gobot"
  "github.com/hybridgroup/gobot/platforms/firmata"
  "github.com/hybridgroup/gobot/platforms/i2c"
)

func main() {
  gbot := gobot.NewGobot()

  firmataAdaptor := firmata.NewFirmataAdaptor("firmata", "COM4")
  mpu6050 := i2c.NewMPU6050Driver(firmataAdaptor, "mpu6050")

  work := func() {
    gobot.Every(300*time.Millisecond, func() {
      fmt.Println("Accelerometer", mpu6050.Accelerometer)
      fmt.Println("Gyroscope", mpu6050.Gyroscope)
      fmt.Println("Temperature", mpu6050.Temperature)
    })
  }

  robot := gobot.NewRobot("mpu6050Bot",
    []gobot.Connection{firmataAdaptor},
    []gobot.Device{mpu6050},
    work,
  )

  gbot.AddRobot(robot)

  gbot.Start()
}

sgilani, 2015-10-26:
By the end i solved my problem in another way for get the values in go:

1.: Adruino Uploaded the MPU6050 RA Example
2.: Done a conection with serial to get the values

package main

import (
         serial "github.com/tarm/serial"
         "log"
				 
)

func main() {
         conf := new(serial.Config)
         conf.Name = "COM4"
         conf.Baud = 57600
				 conf.ReadTimeout = 0

         sc, err := serial.OpenPort(conf)
         if err != nil {
                 log.Fatal(err)
         }

         buf := make([]byte, 1024)

         for {
                 n, err := sc.Read(buf)
                 if err != nil {
                         log.Fatal(err)
                 }
                 log.Printf("%s\n", buf[:n])
         }
}

sgilani, 2015-10-26:
UPDATE:

In Release 8.2 the MPU6050 Driver is Broken. In Version 8.1 it works more or less, the Values are not refreshed, after removing in github.com/hybridgroup/gobot/platforms/i2c/mpu6050_driver.go this part:

	// setSleepEnabled
	if err = h.connection.I2cWrite([]byte{MPU6050_RA_PWR_MGMT_1,
		MPU6050_PWR1_SLEEP_BIT,
		0}); err != nil {
		return
	}

All looks fine and i have all my values. Later i will make a diff between 8.2 and 8.1 to see what is broken and why.

Maybe for the moment it is helpfull for other users.

deadprogram, 2015-10-26:
Hi, @sgilani 

Can you please try this code in substitute for the `initialize` method?

```
func (h *MPU6050Driver) initialize() (err error) {
	if err = h.connection.I2cStart(mpu6050Address); err != nil {
		return
	}

	// setClockSource
	if err = h.connection.I2cWrite(mpu6050Address, []byte{MPU6050_RA_PWR_MGMT_1,
		MPU6050_PWR1_CLKSEL_BIT,
		MPU6050_PWR1_CLKSEL_LENGTH,
		MPU6050_CLOCK_PLL_XGYRO}); err != nil {
		return
	}

	// setFullScaleGyroRange
	if err = h.connection.I2cWrite(mpu6050Address, []byte{MPU6050_RA_GYRO_CONFIG,
		MPU6050_GCONFIG_FS_SEL_BIT,
		MPU6050_GCONFIG_FS_SEL_LENGTH,
		MPU6050_GYRO_FS_250}); err != nil {
		return
	}

	// setFullScaleAccelRange
	if err = h.connection.I2cWrite(mpu6050Address, []byte{MPU6050_RA_ACCEL_CONFIG,
		MPU6050_ACONFIG_AFS_SEL_BIT,
		MPU6050_ACONFIG_AFS_SEL_LENGTH,
		MPU6050_ACCEL_FS_2}); err != nil {
		return
	}

	return nil
}
```

Please let me know if that works for you!

sgilani, 2015-10-26:
HI deadprogramm,

thanks for help, this is the result after implentation, no values.

Accelerometer {0 0 0}
Gyroscope {0 0 0}
Temperature 0

sgilani, 2015-10-27:
Maybe also helpful for debuging:

This freezes the values:

h.adaptor().I2cWrite([]byte{MPU6050_RA_PWR_MGMT_1,
		MPU6050_PWR1_SLEEP_BIT,
		0})

but when i remove just MPU6050_RA_PWR_MGMT_1 so the part looks like this:

h.adaptor().I2cWrite([]byte{
		MPU6050_PWR1_SLEEP_BIT,
		0})

I get again refreshed values, but some are missing:

Accelerometer {20268 -120 14648}
Gyroscope {-5008 -747 0}
Temperature 0

Gyro part 3 and Temp are lost somewhere... really strange

sgilani, 2015-10-27:
Solved for v.8.1, change this 

const MPU6050_PWR1_SLEEP_BIT = 6

to this

const MPU6050_PWR1_SLEEP_BIT = 0

then we can also use the setSleepEnabled part without troubles, to get the right values:

Accelerometer {16472 -680 1772}
Gyroscope {-3872 406 -13133}
Temperature 35
Accelerometer {17472 408 4456}
Gyroscope {-3888 -3898 -3417}
Temperature 33
Accelerometer {16836 -120 856}
Gyroscope {-3824 -625 238}
Temperature 35

deadprogram, 2015-10-27:
According to the MPU6050 docs I could find:

```
Get sleep mode status. Setting the SLEEP bit in the register puts the device into very low power sleep mode. In this mode, only the serial interface and internal registers remain active, allowing for a very low standby current. Clearing this bit puts the device back into normal mode. To save power, the individual standby selections for each of the gyros should be used if any gyro axis is not used by the application. 
```

So perhaps, a better thing would be to add a new constant `const MPU6050_PWR1_ENABLE_BIT = 0` and use that. I'd love a a PR that did that! :)

sgilani, 2015-10-27:
You are right and i like your idea :) i will change it and share it.

deadprogram, 2015-11-18:
Hi, @sgilani did you ever have a chance to work up a PR with this? :)

commit e5707501 (2016-02-17): Fix #236 & fix #239 by correcting initialization and temperature conversion for MPU-6050
closed 2016-02-17 by deadprogram