hybridgroup/gobot #240: OpenCV Window Crash Windows

sgilani, 2015-10-29:
Hello again,

when i try the examples from go-opencv, i have no troubles the webcam and also the hello.go works fine, but with gobot opencv i think the camera works fine (LED is ON), but the window to show the image from the camera, freezes.. not responding.. i tryed on different pc with different web cam, all with the same results.. window not responding.

Then i removed everything with the camera, so just a simple basic window.. result: window not responding..

I tryed this example on different windows computers, any ideas? 

package main

import (
	cv "github.com/hybridgroup/go-opencv/opencv"
	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/platforms/opencv"
)

func main() {
	gbot := gobot.NewGobot()

	window := opencv.NewWindowDriver("window")
	camera := opencv.NewCameraDriver("camera", 1)

	work := func() {
		gobot.On(camera.Event("frame"), func(data interface{}) {
			window.ShowImage(data.(*cv.IplImage))
		})
	}

	robot := gobot.NewRobot("cameraBot",
		[]gobot.Device{window, camera},
		work,
	)

	gbot.AddRobot(robot)

	gbot.Start()
}

sgilani, 2015-10-29:
I changed the code and for creating the window i am using now the original go-opencv and a cv.WaitKey(0) helps a lot, i think you need ti implement this WaitKey to, then all will be fine.

package main

import (
	cv "github.com/hybridgroup/go-opencv/opencv"
	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/platforms/opencv"
)

func main() {
	gbot := gobot.NewGobot()

	window := cv.NewWindow("Go-OpenCV")
	defer window.Destroy()

	camera := opencv.NewCameraDriver("camera", 0)

	work := func() {
		gobot.On(camera.Event("frame"), func(data interface{}) {
			window.ShowImage(data.(*cv.IplImage))
			cv.WaitKey(0)
		})
	}

	robot := gobot.NewRobot("cameraBot",
		[]gobot.Device{camera},
		work,
	)

	gbot.AddRobot(robot)

	gbot.Start()
}

deadprogram, 2015-10-30:
Hi, @sgilani thanks for looking into this. From reading OpenCV docs, seems like `WaitKey(0)` waits indefinitely until keyboard input occurs. Can you please try `WaitKey(10)` and see if it redraws display correctly for you?

sgilani, 2015-10-30:
Hello,

Yes also with WaitKey(10) all is fine, but we need to fix some other parts. 

Like:

window := cv.NewWindow("Go-OpenCV")
defer window.Destroy()

Here i am using the original part from opencv the gobot part for creating the window crashes. and another point for the Camera which is missing in Gobot Opencv is the camera.Release()  without this every frame stays in the memory and the app crash after the memory is full.

Regards,
Saber

deadprogram, 2015-10-30:
What exact version of opencv are you using? Also which version of Windows?

sgilani, 2015-10-30:
Hi,

i tryed 2.4.2 , 2.4.9 , 2.4.10 and 3.0 with gobot 0.8.2 and 0.10 in differnt combination, all with the same result.

Tryed on Windows 8.1 32bit and 64bit and also on Windows 10 32bit and 64bit all with the same result, i would like to try on Windows 7 but i have no active install anymore.

deadprogram, 2015-10-30:
OK, so if I get it working correctly with any of those versions, sounds like we should be good. I'll get my Windows 8.1 VM unpacked.

sgilani, 2015-10-30:
Ohh yes please when you got it working, share with me your setup details so i can get it working with gobot. :) Thanks

deadprogram, 2017-10-23:
The previous OpenCV implementation in Gobot has now been replaced by GoCV (https://gocv.io) in this release https://github.com/hybridgroup/gobot/releases/tag/v1.7.0

I am going to close this issue, please reopen if needed.

closed 2017-10-23 by deadprogram