hybridgroup/gobot #241: Reading from Arduino analog pin A5

fjukstad, 2015-10-29:
Hey,

I'm trying to read some input on analog pin A5 from an Arduino Uno, but it fails with a `index out of bounds` at `github.com/hybridgroup/gobot/platforms/firmata/firmata_adaptor.go:206`. This is how I get things going: 

```
gbot := gobot.NewGobot()
firmataAdaptor := firmata.NewFirmataAdaptor("firmata", "/dev/cu.wchusbserial1420")
sensor := gpio.NewAnalogSensorDriver(firmataAdaptor, "sensor", "5")
work := func() {
		gobot.On(sensor.Event("data"), func(data interface{}) {
			fmt.Println(data) 
		})
}
robot := gobot.NewRobot("myrobot",
		[]gobot.Connection{firmataAdaptor},
		[]gobot.Device{sensor},
		work,
	)
gbot.AddRobot(robot)
gbot.Start()
```

Any idea of what might be wrong? It works fine to read analog input from pin A0 and A1. 

Cheers!

sgilani, 2015-10-30:
You are using firmata and A4 & A5 is by default for I2C, open the StandardFirmata.ino and disable I2C, then all must be fine i think.

deadprogram, 2015-10-31:
Sounds like this could use to be in the README for the Firmata platform.

deadprogram, 2016-12-23:
This is now part of the docs for the Firmata platform https://github.com/hybridgroup/gobot/tree/master/platforms/firmata#how-to-use

Thanks everyone!

closed 2016-12-23 by deadprogram