hybridgroup/gobot #246: Ultrasonic Sensor

sgilani, 2015-11-11:
Hi,

i have now no ideas anymore, how can a get the distance from a Ultrasonic Sensor and Arduino with Gobot.

Can you give me a Tip, i am stocked here:

//----------AND NOW??--------------//
//This is the Part from a Arduino Sketch
duration = pulseIn(echoPin, HIGH)
//--------------------------------//

I have no idea how to dev. this in Go "pulseIn(echoPin, HIGH)"

package main

import (
	"fmt"
	"time"

	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/platforms/firmata"
	"github.com/hybridgroup/gobot/platforms/gpio"
)

func main() {
	gbot := gobot.NewGobot()

	firmataAdaptor := firmata.NewFirmataAdaptor("firmata", "COM7")
	trigPin := gpio.NewDirectPinDriver(firmataAdaptor, "sensor", "11")
	echoPin := gpio.NewDirectPinDriver(firmataAdaptor, "sensor", "12")
	led := gpio.NewLedDriver(firmataAdaptor, "led", "13")

	work := func() {

		maximumRange := 200
		minimumRange := 0

		duration := 0
		distance := 0

		gobot.Every(100*time.Millisecond, func() {

			trigPin.DigitalWrite(byte(0))
			time.Sleep(2 * time.Microsecond)

			trigPin.DigitalWrite(byte(1))
			time.Sleep(10 * time.Microsecond)

			trigPin.DigitalWrite(byte(0))

			//----------AND NOW??--------------//
			//This is the Part from Arduino Sketch
			duration = pulseIn(echoPin, HIGH)
			//--------------------------------//

			//Calculate the distance (in cm) based on the speed of sound.
			distance = duration / 58

			if distance > maximumRange || distance < minimumRange {
				/* Send a negative number to computer and Turn LED ON
				to indicate "out of range" */
				fmt.Println("-1")
				led.On()
			} else {
				/* Send the distance to the computer using Serial protocol, and
				turn LED OFF to indicate successful reading. */
				fmt.Println(distance)
				led.Off()
			}

		})

	}

	robot := gobot.NewRobot("makeyBot",
		[]gobot.Connection{firmataAdaptor},
		[]gobot.Device{trigPin, echoPin, led},
		work,
	)

	gbot.AddRobot(robot)

	gbot.Start()
}

deadprogram, 2015-11-20:
Well, in that case I am closing this issue :sob: 

All joking aside, I think this is a duplicate of https://github.com/hybridgroup/gobot/issues/152 

Bottom line is I need to work on this!

billettc, 2016-12-21:
Here your solution ...

`package main

import (
	"gobot.io/x/gobot/platforms/raspi"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot"
	"time"
	"fmt"
)

func main() {
	r := raspi.NewAdaptor()

	trigPin := gpio.NewDirectPinDriver(r, "11")
	echoPin := gpio.NewDirectPinDriver(r, "12")
	led := gpio.NewLedDriver(r, "7")

	work := func() {

		gobot.Every(1 * time.Second, func() {

			println("Starting probing ")
			led.Toggle()

			trigPin.DigitalWrite(byte(0))
			time.Sleep(2 * time.Microsecond)

			trigPin.DigitalWrite(byte(1))
			time.Sleep(10 * time.Microsecond)

			trigPin.DigitalWrite(byte(0))
			start := time.Now()
			end := time.Now()

			for {
				val, err := echoPin.DigitalRead();
				start = time.Now()

				if (err != nil) {
					println(err)
					break
				}

				if val == 0 { continue }

				break
			}

			for {
				val, err := echoPin.DigitalRead();
				end = time.Now()
				if (err != nil) {
					println(err)
					break
				}

				if val == 1 {
					continue
				}

				break
			}

			duration := end.Sub(start)
			durationAsInt64 := int64(duration)
			distance := duration.Seconds() * 34300
			distance = distance / 2 //one way travel time
			fmt.Printf("Duration : %v %v %v \n", distance, duration.Seconds(), durationAsInt64)
		})
	}

	robot := gobot.NewRobot("makeyBot",
		[]gobot.Connection{r},
		[]gobot.Device{trigPin, echoPin, led},
		work,
	)

	robot.Start()
}`

willmendil, 2020-12-28:
Have there been any updates on this mater?

gen2thomas, 2023-01-29:
An alternative sensor would be the SRF08 I2C sensor, see #327

gen2thomas, 2023-11-10:
I'm going to close this issue, because there will be no good solution for this requirement. Please see the referenced issues for better alternatives.

closed 2023-11-10 by gen2thomas