hybridgroup/gobot #247: Restrictive I2c modelization

abourget, 2015-11-20:
Hi there,

I'm trying to implement the Edison miniboard and I see that the I2cStart, I2cRead and I2cWrite functions are all attached to the Adaptor. However, the Edison miniboard has 2 I2C controllers..

The mraa exposes the different I2C controllers using an `i2c_context`.. and you can have many such contexts (or `devices`) for a given board.

Also, it's not possible for a user to select a certain I2C device, the Adaptor has its default and that's it.

Could you imagine reworking this design ?  Would such a proposition be mergeable ?

abourget, 2015-11-20:
I'm looking at many things in `mraa` that we could use.. is the goal here to have a clean rewrite of those features ? or could we imagine using `mraa` (cgo)  at some point ?

deadprogram, 2015-11-20:
You could be easily done in Gobot by extending via the needed Linux files paths to the other i2c buses in the intel-iot-/edison platform. I do not think you need to drag in all of MRAA just for that.

MRAA has the limitations of blocking, and cgo also has many limitations. The current implementation has the advantage of being pure Go.

That said, there are platforms in Gobot that use cgo wrappers, such as opencv. Nothing prevents a parallel implementation that wraps mraa and/or upm. But the pure Golang approach has many benefits.

abourget, 2015-11-20:
Yeah.. I'd want to keep pure go too.. what about adding a layer behind I2cStart / I2cRead / I2cWrite .. which would use a `DefaultI2cBus`.. and add the notion of an I2cBus .. I2cStart for all platforms that have only one would continue to work, creating a Adaptor.I2cBus .. but some adaptors could expose the different `I2cBus`es as a list or map or something..

This would keep retrocompatibility, keep the easy way like it is, but allow for adaptors with many buses.. and give you a way to pass the I2cBus around if you want..

What do you think ?

deadprogram, 2015-11-20:
That sounds like a good approach towards a first implementation to me. If you work on this, and you want feedback, submit a PR early with comment that it is Work In Progress and not ready to merge. Or just do it! :)

abourget, 2015-11-24:
I wanted to implement the LSM9DS0 first.. and see what would be needed.

You can take a look here: https://github.com/abourget/gobot/commits/master

The i2c interface isn't very slick.. but I do feel it should be more furnished.. It *is* a re-implementation
of what we have in C in the kernel or in mraa.. and I agree it's useful that it be in Go, but I think we'd need to make it a more generic solution.. perhaps that other projects could use.. but that would be more complete.

What do you think ?  Most boards use the /dev/i2c-X bus.. only the firmara doesn't use that.. I'm not sure what messages can go through there.. but it'd be neat to have a solid /dev/i2c-X interface.

I'd like to model an I2CDevice with its address (and keep Go conventions of caps for acronyms), an `I2CBus` to represent `/dev/ì2c-X`..  and a few read/write methods.. this would make porting libs a lot easier..  Also, it'd be nice to be able to decouple the Adaptor from an I2CBus.. and perhaps get a DefaultI2CBus() on the Adaptor .. or embed it, so the Adaptor is also an I2CBus (with a default bus?)

there.. my 2 cents and a half ! good night

ryanhatfield, 2016-02-18:
If there's no objections, I'll likely take a crack at this in the next week or so.

My plan is to convert the [i2cStarter.i2cStart](https://github.com/hybridgroup/gobot/blob/dev/platforms/i2c/i2c.go#L24) method to accept a params object, something like below:

```go
type i2cParams struct {
  // Address would help retain existing functionality
  Address int
  // BusID would default to existing behavior in platforms with multiple i2c bus
  //   but be configurable now for platforms like the Intel Edison that have multiple i2c bus available
  BusID int
}
```

I need to be able to use the `/dev/i2c-2` bus instead of the `/dev/i2c-6` bus that's currently hard coded. What's the best way to do this, so I can have it pulled back into the main repo? :smile: 

@deadprogram or @abourget thoughts?

deadprogram, 2016-02-18:
The only possible problem would be with breaking compatibility with other platforms. Since we're really only talking about 1 more optional param, perhaps a variadic param would be a better idea.

abourget, 2016-02-19:
Didn't I push any proposal ? I remember working on this quite a bit.. maybe
it's only in my fork..

deadprogram, 2016-02-19:
Looks like you really did make progress on this, @abourget sorry I did not look into it sooner. @ryanhatfield this does appear to solve some of your needs, although I think you have a good idea in making the code modular to support multitudes of different breakout boards.

How can we put this all together into something coherent with the latest dev branch? Seems to me there is some good work here, we just need to put it all together.

deadprogram, 2016-02-27:
Hi, @abourget & @ryanhatfield anything I can do to help unify your respective work? :busts_in_silhouette:

deadprogram, 2016-10-07:
As mentioned here https://github.com/hybridgroup/gobot/issues/317#issuecomment-252161858 after the current refactoring effort merges, I'll *also* be turning my attentions back to this issue. Likewise, if there has been any progress, or further ideas, please share. Thanks everyone!

Hadlock, 2016-10-22:
Merging in lsm9ds0 would be fantastic, currently using @abourget 's fork for my project due to support for this device

deadprogram, 2016-12-23:
Now that version 1.0 is out, I'd like to help out further with this effort.

erkkah, 2017-01-19:
I spent a little time on this, since I needed to get better control of i2c when working on drivers for the Adafruit TSL2561 and DRV2605L lux meter and haptic controller.

I incorporated some of the changes from @abourget to sysfs/i2c_device.go and added a layer in drivers/i2c/i2c.go to use multiple buses.

I kept the old interface in parallel, but an added bonus is that the new interface is leaner on each adaptor as well.

https://github.com/erkkah/gobot/commits/better-i2c

deadprogram, 2017-02-08:
The code from https://github.com/hybridgroup/gobot/pull/364 is now in the `dev` branch. I'd love to hear any feedback from anyone about it.

Next step will be another set of changes intended to expose to developer the ability to specify exactly which bus to use for a particular i2c device.

deadprogram, 2017-02-16:
All of the work I referred to has now been released as part of v1.2.0 thanks again everyone!

closed 2017-02-16 by deadprogram