hybridgroup/gobot #248: Gobot spawns thousands of threads causing it to crash everytime

osiloke, 2015-12-15:
Not sure why robot spawns that many threads (assuming, multiple goroutines are running), but it seems to be spawning threads exponentially. I'm using the current master on OS X and connecting to an arduino and just blinking the led (based on the example).
```
func setupArduino() *gobot.Robot {
	//Arduino bot
	device := findArduino()
	arduino := firmata.NewFirmataAdaptor("arduino", device)
  statusLed := gpio.NewLedDriver(arduino, "led", "13") 

	work := func() {
    gobot.Every(1*time.Second, func() {
      statusLed.Toggle()
    })
  }

  return gobot.NewRobot("arduinoExpansion",
    []gobot.Connection{arduino},
		[]gobot.Device{statusLed},
    work,
  )
}
func main() {
  gbot := gobot.NewGobot()
  gbot.AddRobot(setupArduino())

  gbot.Start()
}
```

osiloke, 2015-12-15:
reinstalled gobot, and it stopped spawning infinite goroutines. I must not have been using the latest master. Although, i now notice that a signal interrupt (ctrl+c) does not shut it down

closed 2015-12-15 by osiloke

trevrosen, 2015-12-15:
@osiloke have a look at the commits referenced in #216 and you should have a sense of how to get the behavior you're looking for. Thanks for using Gobot!