hybridgroup/gobot #250: LED pin 18 not blinking with raspb

TheInsideMan, 2015-12-29:
Hello, Im trying to get a LED on pin 18 to flash with a raspberry pi 2 with the following code... any help appreciated...
```
gbot := gobot.NewGobot()
r := raspi.NewRaspiAdaptor("raspi")
led := gpio.NewLedDriver(r, "led", "18")

work := func() {
   gobot.Every(1*time.Second, func() {
      led.Toggle()
   })
}

robot := gobot.NewRobot("blinkBot",
   []gobot.Connection{r},
   []gobot.Device{led},
   work,
)

gbot.AddRobot(robot)
gbot.Start()
```

I've got pi-blaster deamon running.

Im able to get the LED flashing using python so I know it isnt a hardware issue.

TheInsideMan, 2015-12-29:
I tried to run $ sudo ./aambr 

This does stop the error printing but still no flashing LED :(

deadprogram, 2015-12-29:
Hi, @TheInsideMan thank you for the photo. Looks to me line you are plugged into pin labeled on board number 12, which maps to Linux BCM18. When programming using Gobot, you just use whatever the board is labelled, and we do the mapping for you, so you code should be using "12" for the pin number. We probably need to make a bigger deal of explaining this on every platform.

TheInsideMan, 2015-12-30:
thank you @deadprogram thats working now! much appreciated. Let take the opportunity to thank you fro you work on Gobot.

closed 2015-12-30 by TheInsideMan

Depado, 2016-03-28:
Hi, yes that sure needs some explanation ! 
I had the exact same issue as you @TheInsideMan, thanks a lot for the clarification @deadprogram :D
(When I say, exact same issue, we have the exact same setup and approximately quite the same code haha)

gherlein, 2017-06-25:
Probably need some error handling if you call gpio.NewLedDriver with a pin that is not a GPIO pin...

nilskuhn, 2018-01-12:
Where do you see labels on the raspberry gpio pins? 
I'm using version 2 B of the pi.

deadprogram, 2018-01-12:
I highly recommend https://pinout.xyz/

nilskuhn, 2018-01-12:
thx, I already found it in the meanwhile