hybridgroup/gobot #255: Support for Chip

hrishikesh195, 2016-01-19:
I was wondering if anybody is planning to add support for the new $9 single-board computer C.H.I.P. (http://www.getchip.com/). If not, I'd be happy to start working on that. I don't have any I2C or PWM devices right now to test any code for that but I can submit a pull request for basic GPIO digital read/write functionality to get the ball rolling.

deadprogram, 2016-01-19:
I'd love to see this. I have a couple of them, but have not had a chance to really dig into cross-compiling golang for it yet.

hrishikesh195, 2016-01-31:
I have coded up the support for GPIO and I2C for the CHIP. The GPIO (digital read/write) seems to be working fine but I'm having trouble with I2C. The device I'm using is a Wii Nunchuk. When I start the code the following error floods the screen -

```
2016/01/31 22:15:53 Event does not exist
2016/01/31 22:15:53 Event does not exist
2016/01/31 22:15:53 Event does not exist
2016/01/31 22:15:53 Event does not exist
2016/01/31 22:15:53 Event does not exist
...
...
```

To see if this was something specific to the CHIP, I tried to run it on a Raspberry Pi but the same thing happened. Here's the raspi example code -

```go
package main

import (
	"fmt"

	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/platforms/i2c"
	"github.com/hybridgroup/gobot/platforms/raspi"
)

func main() {
	gbot := gobot.NewGobot()

	raspiAdaptor := raspi.NewRaspiAdaptor("raspi")
	wiichuck := i2c.NewWiichuckDriver(raspiAdaptor, "wiichuck")

	work := func() {
		gobot.On(wiichuck.Event("joystick"), func(data interface{}) {
			fmt.Println("joystick", data)
		})

		gobot.On(wiichuck.Event("c"), func(data interface{}) {
			fmt.Println("c")
		})

		gobot.On(wiichuck.Event("z"), func(data interface{}) {
			fmt.Println("z")
		})
	}

	robot := gobot.NewRobot("chuck",
		[]gobot.Connection{raspiAdaptor},
		[]gobot.Device{wiichuck},
		work,
	)

	gbot.AddRobot(robot)

	gbot.Start()
}
```

Using the command line tools (i2cset and i2cdump) I'm able to read bytes off of the nunchuk on both the CHIP and the Pi -

```
# i2cset -y 0 0x52 0x40 0x00
# i2cset -y 0 0x52 0x00
# i2cdump -y -r 0-5 0 0x52 c
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 7e 70 8e 4b 73 7b                                  ~p?Ks{
```

Are there any known issues with the Wii Nunchuk code in gobot? I'm using the code from the latest `dev` branch.

deadprogram, 2016-02-01:
Hi, @hrishikesh195 I have not been using my wiichuck for so long, it is certainly possible there is some problem with the Gobot driver. Do you have any other i2c devices to test with? Perhaps you can submit a preliminary PR, and I can setup my CHIP to test with.

hrishikesh195, 2016-02-01:
Unfortunately I don't have any other I2C device right now. I have submitted a PR. It would be great if you could test it with some other I2C device. Thanks.

deadprogram, 2016-02-03:
Hi, @hrishikesh195 took a little longer to test things, but I was able to use my GroveLCD https://vine.co/v/iJ6VwDqB737

Seems like we need to open a separate issue on the wiichuck. In the meantime, I am merging this, thank you!

hrishikesh195, 2016-02-05:
Great! Thanks.

I looked into the wiichuck problem a little bit more. Let me open a new issue and add my findings there.

Also, I had tried the forums to get more details on PWM (here's the [post](https://bbs.nextthing.co/t/chip-pwm-output/1474/9?u=hrishikesh195)) but didn't have much luck there. Let me see if directly emailing them helps.

deadprogram, 2016-02-05:
There is a Node.js module from @sandeepmistry that looks like it may have run into the same blocker. I added a GH issue on that repo, just so we can all share info on implementation. https://github.com/sandeepmistry/node-chip-io/issues/1

deadprogram, 2016-02-18:
This was released, closing. Thanks again @hrishikesh195

closed 2016-02-18 by deadprogram