hybridgroup/gobot #256: Stepper motor

SellJamHere, 2016-01-27:
Does the current roadmap include support for stepper motors? If not, I'll take a crack at implementing it.

deadprogram, 2016-01-28:
Hi, @SellJamHere we'd love your PR adding support for steppers. Please and thank you!

deadprogram, 2016-02-27:
Hi, @SellJamHere ever have a chance to work on this? :) We'd still love your PR, if you have.

npotts, 2016-07-29:
So Im giving this a shot.  Before I start getting buried in details and state charts - A few questions:
  - Should there be a "platforms/motors" category or just cram everything in "platforms/gpio"?
  - In order to fulfill the gobot.Driver interface, Ill need to implement the gobot.Connection interface.  Do you have any other places where multiple pins are marshaled into a single robot.Connection (eg i2c, spi, etc) so I can see what would be a reasonable approach?

deadprogram, 2016-07-29:
Hi, @npotts 

A couple answers:

- Probably should go into GPIO, since that is the required interface to be able to support s stepper motor controller.
- The RGB Led is the closest one to what you are doing, I think. Check it out here: https://github.com/hybridgroup/gobot/blob/dev/platforms/gpio/rgb_led_driver.go

Thank you!

npotts, 2016-07-30:
Hey @deadprogram 

Im not ready for a pull request as Ive got to mock this mess up and actually make some motors move, but Ive stubbed out the logic [on my fork](https://github.com/npotts/gobot/blob/dev/platforms/gpio).  My question is more asking if [this](https://github.com/npotts/gobot/blob/dev/platforms/gpio/unipolar-stepper.go) the right idea?  Ive never used a stepper in "go" mode, but always in more of a "step 230 times to the left" fashion.

Ravenwater, 2016-08-25:
Is there appetite to design a proper abstraction for the concept of motor control so that both stepper as well as DC+Encoder solutions can be used to accomplish motion control? The higher level abstractions would be valuable for simplifying/unifying GN&C solutions, so it would be good to build this into the overall structure of Gobot. As motion control requires the geometry of the drive configuration this is a non-trivial task. Hierarchical GN&C can be used to subsystem the motion control so that you can used closed loop motor control algorithms to approximate positioning.

From a system design point of view the idea that stepper motors are like LEDs is not helping the architecture of the software.

deadprogram, 2016-08-26:
Well, these are really 2 different activities (simple motor control vs. coordinated steppers). 

For sure, some higher level abstraction around stepper motor control would be very useful. For example, something that would allow for an "easy" inverse kinematics API.

deadprogram, 2017-03-24:
Hello, all. I'm getting ready to devote some attention again to this subject. I will likely implement the simplest possible 4-pin stepper based on the ULN200XA first, and then look at any abstractions on top of it.

Any thoughts on this from anyone here?

coding-yogi, 2017-11-29:
any updates on this? If not, I would like to contribute as well

deadprogram, 2017-11-29:
Hi @coding-yogi I have not had tome to work on this. However, I think that perhaps someone has done so in a fork? In any case, we'd love to have a contribution!

coding-yogi, 2017-11-29:
I am planning to replicate [this](https://github.com/mikedamage/wpi-stepper/blob/master/es6/lib/stepper.js)

deadprogram, 2017-11-29:
That looks good to me!

coding-yogi, 2017-12-02:
@deadprogram , please check [this](https://github.com/coding-yogi/gobot/commit/0fa30aeb62b6b777d70df00c03b367f76f44bed6). Works fine for me

deadprogram, 2017-12-02:
That looks great to me, @coding-yogi

Please submit a PR at your convenience, and thank you!

deadprogram, 2017-12-02:
The new stepper created by @coding-yogi is now in the `dev` branch. Please check it out, everyone.

I will close this issue once the code makes it into the next release.

deadprogram, 2017-12-21:
Stepper motor driver was included in v1.8.0 so I am now closing this issue. Please reopen if needed, and thank you to everyone especially @coding-yogi for the PR.

closed 2017-12-21 by deadprogram