hybridgroup/gobot #258: 1wire protocol

pbdeuchler, 2016-02-01:
I'm currently trying to use a DS18B20 temperature sensor with an arduino and gobot but I can't find any golang implementations of the 1wire protocol. Has anyone used one of these sensors in go? If not does anyone know where the protocol is documented so I can implement it in go?

pbdeuchler, 2016-02-01:
FYI there is support with [arduino configurable firmata](https://github.com/firmata/ConfigurableFirmata/blob/master/src/OneWireFirmata.cpp), if that would be easier to write a go shim for

ryanhatfield, 2016-02-18:
As I remember, the 1 wire protocol is fairly slow and easy to implement if you can time things right.

Here's the [wiki article about the 1 wire protocol](https://en.wikipedia.org/wiki/1-Wire), and they explain the timing structure there if you wanted to implement it in go.

Also, here's a [1 wire protocol for RaspberryPi](https://github.com/aqua/raspberrypi) written in go, and looks simple enough.

deadprogram, 2016-02-18:
Hi, @ryanhatfield thanks for those links... I'll dig in soon.

pbdeuchler, 2016-02-18:
Seconded on the thanks @ryanhatfield... I might implement this as well @deadprogram since I'm currently in need of it for a project. Just glad I don't have to reverse engineer the arduino library.

pbdeuchler, 2016-02-18:
On a quick inspection it looks like that go library is just using the official driver to read/write from a device file, which we won't have the luxury of doing since AFAIK gobot requires direct control due to the different platforms

deadprogram, 2016-02-18:
The solution to that in Gobot, is to define an interface for 1Wire, and then adaptors for different platforms can implement that interface in their own way. For Raspi & other Linux systems probably using driver, for Arduino targeting ConfigurableFirmata, etc.

gen2thomas, 2025-02-16:
fixed in v2.5.0

closed 2025-02-16 by gen2thomas