hybridgroup/gobot #262: Errors on Intel Edison, Sans-Arduino

ryanhatfield, 2016-02-18:
I have an Intel Edison without the Arduino Development board, and the [edison_adapter](https://github.com/hybridgroup/gobot/blob/dev/platforms/intel-iot/edison/edison_adaptor.go#L199) is looking for pins that don't exist.

According to this [Guide on Multiplexing](http://www.emutexlabs.com/project/215-intel-edison-gpio-pin-multiplexing-guide) with the Intel Edison, some of the pins with hard references in code won't exist if you're using the Intel Edison without the pin expanders.

Removing any [GPIO Pins larger than 200](https://github.com/hybridgroup/gobot/blob/dev/platforms/intel-iot/edison/edison_adaptor_test.go#L64) should help show what would happen if those pins are missing, and any pins over 200 are added by the i2c port expander. (as I understand it).

I'll be working to remove the pins that are causing issues right now, but is there any larger issues I'll run into with these pins being unavailable?

deadprogram, 2016-02-18:
Hi, @ryanhatfield thanks for pointing this out, you are correct that the current implementation makes some assumptions about the Arduino-compatible breakout board being present. 

Are you using the mini-breakout board, the Sparkfun board, or something else entirely?

One possible solution that comes to mind, is pulling out the shared Edison muxing functions into a separate file, and then creating new adaptors for each of the specific breakout boards.

What do you think?

ryanhatfield, 2016-02-18:
I think having the _Arduino_ section as it's own _Platform_ or device would be another way to solve this. I would think since the i2c expansion is being added, a "i2c port expander" device could be added, and the edison arduino might have a platform specific version to handle the unique way it gets added?

I have the mini-breakout board, the Sparkfun Edison blocks, and a fully custom board at my disposal, so pure abstraction of the Intel Edison's base functionality is a requirement for me. 

Also, I do a certain amount of Kernel configuration for some pins / registers, I will likely look into a way of dynamically or statically defining the pin mappings (if different than default) and that could also facilitate the additional pins available on the Arduino variant.

ryanhatfield, 2016-02-18:
Also, with the speed I'm moving at with some of these things, I might not take the time to back port functionality into all platforms right away. I might just keep a branch open on my fork to hold these modifications and when someone (possibly me) has time, they can be fully integrated or modified to be pulled back upstream.

deadprogram, 2016-02-18:
I also have mini-breakout board + Sparkfun blocks, so I can help as well, as far as either covering, or at least documenting any gaps. This sounds great!

devfubar, 2016-10-06:
Is there any update on this?  I just ran into exactly this issue.....

deadprogram, 2016-10-07:
As mentioned here https://github.com/hybridgroup/gobot/issues/317#issuecomment-252161858 after the current refactoring effort merges, I'll be turning my attentions back to this. If there has been any progress, or further ideas, please do tell. Thanks!

devfubar, 2016-10-07:
Thanks, was just interested to see if this was still on the agenda since the last comment was a while ago.

I'm under a bit of time pressure for my project to create a POC.  I really want to use gobot so is there anything I can do as a work around?  I have tried commenting out the offending parts such as [this section](https://github.com/hybridgroup/gobot/blob/5442e0bd8b0c909b552677b2790b4eddbb7e7e48/platforms/intel-iot/edison/edison_adaptor.go#L189-L197) as @ryanhatfield mentioned however I suspect this has consequences as I see some weird behaviour.

deadprogram, 2016-10-07:
@devfubar which board are you using?

devfubar, 2016-10-07:
Sparkfun with GPIO block

deadprogram, 2016-10-07:
OK, I have a WIP branch here https://github.com/hybridgroup/gobot/tree/feature/edison-other-boards that I cannot test myself due to not having the HW right now. However, it might work :)

To use, you need to change the expansion board using `SetBoard()` to something other than `"arduino"` like this:

```go
board := edison.NewAdaptor()
board.SetBoard("miniboard")
```

If anyone can give this a try with a Miniboard or Sparkfun expansion board, that would be awesome!

devfubar, 2016-10-07:
Thanks @deadprogram .  I'll give it a bash and see how far I get...

deadprogram, 2016-10-21:
All of these changes have been merged into the current `dev` branch https://github.com/hybridgroup/gobot/tree/dev please take a look and try it out!

Note that there have been some other substantial changes to Gobot that will be coming soon as part of the upcoming 1.0 release that are also part of the dev branch.

devfubar, 2016-10-21:
OK, I'll probably give it a shot over the weekend.  I also want to add an Edison pinmap file and a switch [here](https://github.com/hybridgroup/gobot/blob/54d9b161417e4b329dc27070d460936be35ba444/platforms/intel-iot/edison/edison_adaptor.go#L96) to handle the arduino, miniboard and SparkFun GPIO block.
The reason for the new pinmap is because the miniboard one has differant labels to what people with a SparkFun would expect eg ["J19-9"](https://github.com/hybridgroup/gobot/blob/54d9b161417e4b329dc27070d460936be35ba444/platforms/intel-iot/edison/miniboard_pinmap.go#L256) should be just "14".

How does that sound @deadprogram ?

deadprogram, 2016-10-21:
That sounds amazing, exactly what it needs. Thanks!

deadprogram, 2016-11-01:
The `dev` branch now has all of these changes, any interested parties should please give it a try. Reopen this issue if needed, thank you very everyone who contributed to the discussion and the resulting code.

closed 2016-11-01 by deadprogram