hybridgroup/gobot #272: Intel Edison read i2c register by address

allanglen, 2016-03-17:
I am writing a driver for the MPU9150 IMU.  I am attempting to read a byte from a specific register and am always getting back the value from the first register on the device.

The sysfs i2c implementation sets the command byte to 0 for all reads:
https://github.com/hybridgroup/gobot/blob/dev/sysfs/i2c_device.go#L103

Per the documentation here (https://www.kernel.org/doc/Documentation/i2c/smbus-protocol), an I2C_SMBUS_I2C_BLOCK_DATA operation uses the command byte to set the starting register for the read but that is always set to 0.

An I2C_SMBUS_READ_BYTE operation will use the register from a previous write but not I2C_SMBUS_I2C_BLOCK_DATA.

Are all drivers that currently implemented just happening to be reading from the first register or am I missing something obvious?

deadprogram, 2016-03-18:
Looks like this is an oversight that does not impact you if you are using an i2c adaptor that does not support the block data reads.

The solution is probably use the first byte of `b` as the register for that call. What do you think?

deadprogram, 2016-12-06:
Hi @allanglen looked at this with fresh eyes, and I see that the `Write()` command is doing the right thing: https://github.com/hybridgroup/gobot/blob/dev/sysfs/i2c_device.go#L128

I am going to make a similar change to `Read()`, would you be able to test this out on your hardware config?

deadprogram, 2017-01-02:
Hi @allanglen this should now be fixed in the `dev` branch.

commit c432f50a (2017-01-02): sysfs: Should fix #272 by using first byte of data as command register for I2C reads
commit dcc63a1f (2017-01-14): sysfs: Should fix #272 by using first byte of data as command register for I2C reads
closed 2017-01-09 by deadprogram