hybridgroup/gobot #275: Support for new cape manager in Linux 4.x on Beaglebone Black

dsockwell, 2016-04-03:
Since the Beaglebone cape manager has been moved to kernel mainline, some parts of it have changed. The existing beaglebone GPIO driver works great on Linux 3.8 as delivered with the Wheezy images for bbb, but users might not know to downgrade in order for gobot to work.

The best documentation/example I've been able to find on the difference is at https://github.com/cdsteinkuehler/beaglebone-universal-io/blob/master/config-pin - the script branches based on the existence of a certain part of the 4.x implementation and configures pins a little differently.

I am working on my first Go project, but I may come up with a PR for 4.x kernels in the future. Someone else already more proficient with Go would probably not have much trouble making an alternate driver or putting the alternate behavior in the current one.

deadprogram, 2016-04-03:
Hi @dsockwell thanks for creating this issue. I'm still running Wheezy myself on my various BBBs, so I've not run into this yet.

dsockwell, 2016-04-03:
A big part of the difference is that the current GPIO driver calls on cape files that are not present in 4.x. It looks like individual pin configuration could be worked into the individual pin drivers as opposed to the overarching GPIO driver for the BBB, but also approximating the PWM configuration at least could go into the connect code to keep things laid out the same. This is a plug-and-play framework so i wouldn't be too upset about dictating pin configurations for the sake of keeping it uniform.

nevers, 2016-08-07:
I just upgraded to Debian Jessy and I stumbled on this very same issue. Really sad that its still broken... Is anyone still working on this? I'll try and an have a look myself too.

deadprogram, 2016-11-26:
Hi everyone. Please check out the https://github.com/hybridgroup/gobot/tree/feature/beaglebone-4 branch which should work with the latest Debian, as well as with the older kernel versions if needed.

deadprogram, 2016-11-30:
Here is a PR for any comments: https://github.com/hybridgroup/gobot/pull/342

deadprogram, 2016-12-03:
This has been merged into the `dev` branch. Thanks everyone!

closed 2016-12-03 by deadprogram