hybridgroup/gobot #276: Add GoPiGo Support

DexterInd, 2016-04-14:
Is it pssible to add GoPiGo support?  https://github.com/DexterInd/GoPiGo/tree/master/Software

deadprogram, 2016-04-15:
Hi @DexterInd I was reviewing your code/hardware. I see you have a daughterboard which is an i2c device. Seems like the slick route is to use the Gobot raspi platform and its i2c support, to create a new platform for GoPiGo. This new platform would implement an adaptor with the standard Gobot interfaces:

```
var _ gobot.Adaptor = (*GoPiGoAdaptor)(nil)

var _ gpio.DigitalReader = (*GoPiGoAdaptor)(nil)
var _ gpio.DigitalWriter = (*GoPiGoAdaptor)(nil)
var _ gpio.AnalogReader = (*GoPiGoAdaptor)(nil)
var _ gpio.PwmWriter = (*GoPiGoAdaptor)(nil)
var _ gpio.ServoWriter = (*GoPiGoAdaptor)(nil)
```

This would allow using any of the current GPIO devices. 

The new GoPiGo Gobot platform would also want to additionally implement some Gobot devices specific to the GoPiGo's built-in hardware, such as the IRReceiver. These devices would ideally define their own interfaces. In the IRReceiver's case something based on `io.Reader` might be cool. 

Does all this make sense?

deadprogram, 2016-05-13:
Hi @DexterInd I was looking thru my gear, and seems I was one of your Kickstarter backers, as I have one of your boards. Do please let us know anything needed to help out.

DexterInd, 2016-05-16:
Hey @deadprogram , we are going to put this on our list of projects, sorry for the slow response here.  Just need to find the time.  If anyone is interested in developing on this or working on this though, we're happy to send along a GoPiGo.

deadprogram, 2017-10-23:
GoPiGO3 support has just been released as part of https://github.com/hybridgroup/gobot/releases/tag/v1.7.0 so I am now going to close this. Thank you everyone who contributed!

closed 2017-10-23 by deadprogram