hybridgroup/gobot #277: MQTT reconnect?

tlrobinson, 2016-04-14:
It looks like MQTT doesn't support automatic reconnection (or a way to set custom ClientOptions): https://github.com/hybridgroup/gobot/blob/dev/platforms/mqtt/mqtt_adaptor.go#L92

This seems like it would be a very useful thing. Is there a reason it's not exposed?

deadprogram, 2016-04-15:
Him @tlrobinson we for sure need a way to set all the various `ClientOptions`. A PR that added this would be very appreciated... :)

tlrobinson, 2016-04-16:
(Sorry, accidentally closed this)

Happy to submit a PR, just wanted to make sure it was desirable.

I'm guessing even if you set `AutoReconnect` to true it won't automatically re-subscribe to any topics you initially subscribed to, so you should do so in an `OnConnectHandler`. I'm new to gobot, is there an existing pattern for that sort of thing?

deadprogram, 2016-04-17:
The simplest implementation is probably: 

- add a slice to the Adaptor struct to keep track of the current topics. 
- in the Adaptors's `Connect()` method you can hookup the `OnConnectHandler` to a method that subscribes to all of the topics in your list.
- the Adaptor's `On` handler can add new topics to the list, and also calls `Subscribe()` if the connection is already live.
- When a connect or reconnect occurs, it will call the `OnConnectHandler` to subscribe to all the list topics

Does that sound about right to you?

tlrobinson, 2016-04-18:
That makes sense, and would be really nice to use in a lot of cases. I went ahead and implemented this and it seems to work well.

However, in my case I need to ignore the first message my MQTT server sends when I subscribe, so if we transparently re-subscribe I won't be able to know if it's the first message or not (though I suppose I could check to see if it's identical to the previous message). There may be other reasons why someone would want to know if the connection is lost as well, so perhaps we should expose `OnConnect` and `OnDisconnect` events?

deadprogram, 2016-04-18:
That sounds like a very reasonable idea to me as well.

deadprogram, 2016-05-02:
Hi @tlrobinson did you ever have a chance to do anything on this, or do you need any additional assistance? Please let us know if there is anything we can do to help. Thanks.

deadprogram, 2017-01-26:
I was just looking at the code for the Paho MQTT lib, and looks to me like a `reconnect()` will maintain the same subscribe handlers, since all of the clidnt.msgRouter routes are preserved until the `mqtt.Client` is destroyed.

Basically, all we need to do is expose the clientoptions, and let Paho do the rest.

deadprogram, 2017-01-26:
OK this feature is now in the `dev` branch, and I verified it against a remote MQTT broker. I will close this issue next release.

commit 4b8546f8 (2017-01-26): mqtt: fix #277 by adding SetAutoReconnect method to set Paho MQTT client
closed 2017-02-16 by deadprogram