hybridgroup/gobot #281: [RFC] Add support for seven segment display

lyrixx, 2016-05-21:
Hello.

First, thanks a lot for this project. It works very well and it's much more fun to play with Go than C ;)

So I have started to learn all that stuff and I have followed a tutorial. And in the current chapter we have to use a seven segment display. I managed to use / control it with the `NewDirectPinDriver`. 
But a thin abstraction layer could be be awesome to control the display.

What do you think ?

deadprogram, 2016-05-21:
Hi, @lyrixx

That sounds great, we'd love a PR with a `SevenSegmentLedDriver`.

lyrixx, 2016-05-21:
But I have few questions before doing a PR:

I very strict mapping is needed between pins ship, so I'm not sure how to handle that. Maybe with a figure like the one about the [button](https://gobot.io/documentation/drivers/button/)

The other question is about the BCD (Binary code Decimal) Decoder. I did not use it for now and so I used 8 pins. I'm going to see if I have one to test how does it work. But I don't know how to handle this in the code. May be with two Drivers ? or with a flag ?

For now my code looks like this:

```
	displayNumber := func(i int) {
		for _, pin := range pins {
			pin.DigitalWrite(0)
		}

		i = i % 10

		if i == 0 {
			pin2.DigitalWrite(1) // low right
			pin3.DigitalWrite(1) // low middle
			pin4.DigitalWrite(1) // low left
			pin6.DigitalWrite(1) // high left
			pin7.DigitalWrite(1) // high middle
			pin8.DigitalWrite(1) // high right
		} else if i == 1 {
			pin2.DigitalWrite(1) // low right
			pin8.DigitalWrite(1) // high right
		} else if i == 2 {
			pin7.DigitalWrite(1) // high middle
			pin8.DigitalWrite(1) // high right
			pin5.DigitalWrite(1) // middle middle
			pin4.DigitalWrite(1) // low left
			pin3.DigitalWrite(1) // low middle
		} else if i == 3 {
			pin2.DigitalWrite(1) // low right
			pin3.DigitalWrite(1) // low middle
			pin5.DigitalWrite(1) // middle middle
			pin7.DigitalWrite(1) // high middle
			pin8.DigitalWrite(1) // high right
		} else if i == 4 {
			pin2.DigitalWrite(1) // low right
			pin5.DigitalWrite(1) // middle middle
			pin6.DigitalWrite(1) // high left
			pin8.DigitalWrite(1) // high right
		} else if i == 5 {
			pin2.DigitalWrite(1) // low right
			pin3.DigitalWrite(1) // low middle
			pin5.DigitalWrite(1) // middle middle
			pin6.DigitalWrite(1) // high left
			pin7.DigitalWrite(1) // high middle
		} else if i == 6 {
			pin2.DigitalWrite(1) // low right
			pin3.DigitalWrite(1) // low middle
			pin4.DigitalWrite(1) // low left
			pin5.DigitalWrite(1) // middle middle
			pin6.DigitalWrite(1) // high left
			pin7.DigitalWrite(1) // high middle
		} else if i == 7 {
			pin2.DigitalWrite(1) // low right
			pin7.DigitalWrite(1) // high middle
			pin8.DigitalWrite(1) // high right
		} else if i == 8 {
			pin2.DigitalWrite(1) // low right
			pin3.DigitalWrite(1) // low middle
			pin4.DigitalWrite(1) // low left
			pin5.DigitalWrite(1) // middle middle
			pin6.DigitalWrite(1) // high left
			pin7.DigitalWrite(1) // high middle
			pin8.DigitalWrite(1) // high right
		} else if i == 9 {
			pin2.DigitalWrite(1) // low right
			pin3.DigitalWrite(1) // low middle
			pin5.DigitalWrite(1) // middle middle
			pin6.DigitalWrite(1) // high left
			pin7.DigitalWrite(1) // high middle
			pin8.DigitalWrite(1) // high right
			i = -1
		}
	}
```

christingh, 2016-05-21:
BCD (Binary code Decimal) Decoder. I did not use it for now and so I used 8 pins. I'm going to see if I have one to test how does it work. But I don't know how to handle this in the code. May be with two Drivers ? or with a flag ??,,

christingh, 2016-05-21:
(0)
        }

        i = i % 10

        if i == 0 {
            pin2.DigitalWrite(1) // low right
            pin3.DigitalWrite(1) // low middle
            pin4.DigitalWrite(1) // low left
            pin6.DigitalWrite(1) // high left
            pin7.DigitalWrite(1) // high middle
            pin8.DigitalWrite(1) // high right
        } else if i == 1 {
            pin2.DigitalWrite(1) // low right
            pin8.DigitalWrite(1) // high right
        } else if i == 2 {
            pin7.DigitalWrite(1) // high middle
            pin8.DigitalWrite(1) // high right
            pin5.DigitalWrite(1) // middle middle
            pin4.DigitalWrite(1) // low left
            pin3.DigitalWrite(1) // low middle
        } else if i == 3 {
            pin2.DigitalWrite(1) // low right
            pin3.DigitalWrite(1) // low middle
            pin5.DigitalWrite(1) // middle middle
            pin7.DigitalWrite(1) // high middle
            pin8.DigitalWrite(1) // high right
        } else if i == 4 {
            pin2.DigitalWrite(1) // low right
            pin5.DigitalWrite(1) // middle middle
            pin6.DigitalWrite(1) // high left
            pin8.DigitalWrite(1) // high right
        } else if i == 5 {
            pin2.DigitalWrite(1) // low right
            pin3.DigitalWrite(1) // low middle
            pin5.DigitalWrite(1) // middle middle
            pin6.DigitalWrite(1) // high left
            pin7.DigitalWrite(1) // high middle,!

deadprogram, 2016-05-21:
Sorry, I am not fully understanding. Please elaborate on "May be with two Drivers ? or with a flag ?"

christingh, 2016-05-21:
thanks a lot for this project. It works very well and it's much more fun tothnee,  "May be with two Drivers ? or with a flag ?"

lyrixx, 2016-05-21:
Right now, I use 8 wires from the arduino to the seven segment display.
But I saw I can also use a BCD decoder to save some arduino pins. But as I said, I'm a noob with everything ;) 
I have to dig more in this direction to see how BCD works. 
I'm going to try ASAP and I will update this RFC ASAP.

deadprogram, 2016-05-21:
OK, now I think I understand. You are proposing both a `SevenSegmentLedDriver` and also a `BcdLedDriver` that have the same interface, but the implementation uses either 8 pins (for the  `SevenSegmentLedDriver` or 4 pins (for the `BcdLedDriver`)?

deadprogram, 2016-05-21:
Also, you might want to use a map with arrays to represent the different digits, vs. a long conditional statement. Just a small suggestion.

trevrosen, 2016-05-23:
+1 for map to hold this data

trevrosen, 2016-06-03:
Any movement on this feature? Looking forward! Please reach out if you happen to require any general help with Go. 😺

lyrixx, 2016-06-03:
Hello. Sorry. I did not get time to work again on this project :/

lyrixx, 2020-01-30:
I'm closing this issue because I will never find time to contribute to it.
Sorry for the noise

closed 2020-01-30 by lyrixx