hybridgroup/gobot #283: run only one goroutine at a time

nhooyr, 2016-06-06:
Every time a button is pressed, I want to ensure that if a goroutine is already running and handling previous input, the new goroutine exits. I see two ways to do this. One is using a channel with one sized buffer or launching a always running goroutine and sending the button presses to it using a select to make it nonblocking.

Here is option one:

```go
var running = make(chan struct{}, 1)

func work() {
	gobot.On(button.Event("push"), func(data interface{}) {
		log.Print(data)
		select {
		case running <- struct{}{}:
		default:
			return
		}
		// work
		defer func() {
			<-running
		}()
	})
// ... more code
}
```

Option two is simple so I wont provide an example. Which would be better?

nhooyr, 2016-06-07:
I think I realized a small problem with option two. There is no guarantee that the real work goroutine is launched and listening on the channel for input when the button is pressed whereas in option one it is guaranteed. Thus I think option one is the cleaner approach.

nhooyr, 2016-06-23:
I think I realized a small problem with option two. There is no guarantee that the work goroutine is launched and listening on the channel for input when the button is pressed. Thus at the start, it could lose input (unless I add a timeout or something instead of using default) whereas in option one it is guaranteed. Thus I think option one is the cleaner approach.

Ravenwater, 2016-08-25:
Conceptually, what would be the situation you want to avoid:
1- that the initiator has multiple outstanding requests queuing up, or
2- that the target has multiple overlapping state updates?

For #1, the initiator would test the button press state and filter out concurrent presses (it doesn't matter that you don't know if the down stream has started, all you need to know is if you have dispatched the work, at which point it is in play). For #2, the target would filter the state update overlap. Using a single slot channel should be sufficient to never have multiple overlapping state updates). There is one condition that is up to you to define and that is what to do with multiple button presses? Do you want to keep them as actual actionable events, or can you drop them if they overlap? Clearly, dropping them would yield a simpler system.

deadprogram, 2016-08-31:
I've been thinking a lot about this issue since it was opened, and it identified 2 things. One was that the internal architecture of how Gobot handled events needed to be refactored. The second was that we needed to explain a lot better what is happening "under the hood" as well as better allow for "bare metal" style of development using Gobot as a library and not necessarily the entire framework.

https://github.com/hybridgroup/gobot/pull/306 is the result of our current work on the first issue. Events will be now handled using event channels, which makes the concurrency just work how you expect in Golang.

Regarding the documentation of this, we've started an issue here https://github.com/hybridgroup/gobot-site/issues/48 that should also include information such as being discussed in this one.

Let the conversation continue, and thanks!

deadprogram, 2016-10-21:
Hi there.

There has been subtancial changes to how Gobot creates and manages the goroutines used. Please take a look at the `dev` branch to see what we've been doing: https://github.com/hybridgroup/gobot/tree/dev

In short:

- every Robot's Work is now run inside of a separate goroutine that exists for the life of the Robot.
- every Device's On and Once event handlers are each run in a separate goroutine that exists for the life of the Device.
- channels are now used for Events, and the implementations of On and Once use the event channels as their implementation to communicate when things happen.

The `dev` branch has a lot of breaking changes, but I think you will find the syntax is simpler, and the implementation much cleaner and more concurrent by default.

nhooyr, 2016-10-21:
Thank you, I will take a closer look when I get back to using my raspberry pi.

deadprogram, 2016-11-03:
I am going to close this issue now, please reopen if needed. Thanks, everyone!

closed 2016-11-03 by deadprogram