hybridgroup/gobot #287: why is raspberry pi PWM limited between 0 and 255?

nhooyr, 2016-06-11:
The underlying pi-blaster daemon supports 1000 steps. Why only limit PWM write values between 0 and 255?

deadprogram, 2016-07-20:
Hi, @nhooyr the interface only supports that many, to provide compatibility with other interfaces such as Firmata etc. So really no good reason at all.

Perhaps a "PWMScale" method or something like that to allow setting different variations of PWM control. What do you think? 

I'd be happy to accept a PR for the Raspi platform which added this. ;)

deadprogram, 2016-11-04:
I'm going to close this issue due to inactivity, please reopen if needed. Thanks!

closed 2016-11-04 by deadprogram

verrol, 2019-07-31:
Hi,

Please re-open this issue to add some kind of scale method for PWM modules that uses more than 8bits.

Since gpio.DirectPinDriver.PwmWrite() uses a byte value, the max duty cycle on an ESP8266 is ~25%. 

The ESP8266 appears to have a 10-bit resolution, so when specifying 255, the duty cycle is only a fraction of the max. Please see the attached screenshot from my logic analyzer.

The following code was used to test all 4 PWM pings for the ESP8266.

Thanks,

.v

```go
package main

import (
	"flag"
	"log"
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/firmata"
)

/*

 NodeMCU Pin  | GPIO   | Function Pin
--------------+--------+-------
 10           | 12     | PWM0
 13           | 15     | PWM1
 9            | 14     | PWM2
 16           | 4      | PWM3

*/
const (
	pwm0Pin     = "12"
	pwm1Pin     = "15"
	pwm2Pin     = "14"
	pwm3Pin     = "4"
	defaultPort = "10.10.100.175:3030"
)

func main() {
	flag.Parse()

	port := defaultPort

	if len(flag.Args()) == 1 {
		port = flag.Args()[0]
	}

	log.Printf("Using port %v\n", port)

	board1 := firmata.NewTCPAdaptor(port)
	pwm0 := gpio.NewDirectPinDriver(board1, pwm0Pin)
	pwm1 := gpio.NewDirectPinDriver(board1, pwm1Pin)
	pwm2 := gpio.NewDirectPinDriver(board1, pwm2Pin)
	pwm3 := gpio.NewDirectPinDriver(board1, pwm3Pin)

	var speed byte

	work := func() {
		gobot.Every(2*time.Second, func() {
			log.Printf("Setting PWM to %v", speed)
			pwm0.PwmWrite(speed)
			pwm1.PwmWrite(speed)
			pwm2.PwmWrite(speed)
			pwm3.PwmWrite(speed)

			speed ^= 0xFF // toggle between 0% and 100% duty
		})

	}

	robot := gobot.NewRobot("my-robot",
		[]gobot.Connection{board1},
		[]gobot.Device{pwm0, pwm1, pwm2, pwm3},
		work,
	)

	robot.Start()
}
```

[image]
[image]