hybridgroup/gobot #288: Trouble installing Digispark on Mac OS

jish, 2016-06-14:
I'm following the instructions on this page https://gobot.io/documentation/platforms/digispark/ but I am having issues installing the Digispark platform:

    $ brew install libusb
    Warning: libusb-1.0.20 already installed
    $ brew install libusb-compat
    Warning: libusb-compat-0.1.5 already installed
    $ go get -d -u github.com/hybridgroup/gobot/...

And then

    $ go install github.com/hybridgroup/gobot/platforms/digispark
    # github.com/hybridgroup/gobot/platforms/digispark
    In file included from code/go/src/github.com/hybridgroup/gobot/platforms/digispark/littleWire.go:4:
    ./littleWire.h:34:13: fatal error: 'usb.h' file not found
       #include <usb.h>       // this is libusb, see http://libusb.sourceforge.net/
                ^
    1 error generated.

deadprogram, 2016-07-02:
Hi, @jish what version of golang are you using?

jish, 2016-07-02:
It looks like I'm using version `1.6.2`:

    $ go version
    go version go1.6.2 darwin/amd64

deadprogram, 2017-01-30:
I've just pushed a fix for this to the `dev` branch, I will mark as closed once it is released.

Thanks, everyone!

commit b163b5f2 (2017-01-30): digispark: fix #288 by using pkg-config to locate libusb-compat includes
closed 2017-02-16 by deadprogram