hybridgroup/gobot #298: ollie/bb8: panic: runtime error: cgo argument has Go pointer to Go pointer

broady, 2016-07-18:
```
$ go run rover/ollie/1-color.go BB-EFCC
2016/07/17 21:25:43 Initializing Robot ollieBot ...
2016/07/17 21:25:43 Initializing connections...
2016/07/17 21:25:43 Initializing connection ble ...
2016/07/17 21:25:43 Initializing devices...
2016/07/17 21:25:43 Initializing device ollie ...
2016/07/17 21:25:43 Starting Robot ollieBot ...
2016/07/17 21:25:43 Starting connections...
2016/07/17 21:25:43 Starting connection ble...
State: PoweredOn
scanning...
2016/07/17 21:25:44 Unhandled event: xpc.Dict{"kCBMsgId":53, "kCBMsgArgs":xpc.Dict{"kCBMsgArgDeviceUUID":xpc.UUID{0xf6, 0x92, 0xb4, 0x84, 0xa6, 0xe1, 0x47, 0xa, 0x94, 0x5b, 0x8f, 0x1e, 0xb4, 0xef, 0xaf, 0x85}, "kCBMsgArgATTMTU":23}}

Connected Peripheral ID:f692b484a6e1470a945b8f1eb4efaf85, NAME:()
Failed to set MTU, err: Not implemented
2016/07/17 21:25:44 Unhandled event: xpc.Dict{"kCBMsgId":81, "kCBMsgArgs":xpc.Dict{"kCBMsgArgConnectionLatency":0, "kCBMsgArgSupervisionTimeout":175, "kCBMsgArgDeviceUUID":xpc.UUID{0xf6, 0x92, 0xb4, 0x84, 0xa6, 0xe1, 0x47, 0xa, 0x94, 0x5b, 0x8f, 0x1e, 0xb4, 0xef, 0xaf, 0x85}, "kCBMsgArgConnectionInterval":11}}
2016/07/17 21:25:45 Starting devices...
2016/07/17 21:25:45 Starting device ollie...
panic: runtime error: cgo argument has Go pointer to Go pointer

goroutine 35 [running]:
panic(0x40e5560, 0xc420110a50)
	/Users/cbro/go/src/runtime/panic.go:500 +0x1a1
github.com/currantlabs/gatt/xpc._cgoCheckPointer0(0xc42005c724, 0x0, 0x0, 0x0, 0x40d8901)
	??:0 +0x59
github.com/currantlabs/gatt/xpc.valueToXpc(0x40d8900, 0xc420112480, 0x97, 0x40d8900)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:254 +0x3ff
github.com/currantlabs/gatt/xpc.valueToXpc(0x40e2f40, 0xc420110a40, 0x94, 0x40da560)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:269 +0x68b
github.com/currantlabs/gatt/xpc.valueToXpc(0x40f8e00, 0xc420011830, 0x15, 0x40f8e00)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:239 +0x8b2
github.com/currantlabs/gatt/xpc.valueToXpc(0x40e2f40, 0xc4201109c0, 0x94, 0x40da560)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:269 +0x68b
github.com/currantlabs/gatt/xpc.valueToXpc(0x40f8e00, 0xc4200118c0, 0x15, 0xc4200118c0)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:239 +0x8b2
github.com/currantlabs/gatt/xpc.goToXpc(0x40f8e00, 0xc4200118c0, 0x0)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:213 +0x57
github.com/currantlabs/gatt/xpc.(*XPC).Send(0xc4200a00b0, 0x40f8e00, 0xc4200118c0, 0xc4200f7e00)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:23 +0x85
github.com/currantlabs/gatt.(*device).sendCBMsg(0xc4200a0080, 0x42, 0xc420011830)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/device_darwin.go:509 +0x15b
github.com/currantlabs/gatt.(*peripheral).loop.func1(0xc4200a0180, 0xc4200d8000)
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/peripheral_darwin.go:243 +0x119
created by github.com/currantlabs/gatt.(*peripheral).loop
	/Users/cbro/src/gocode/src/github.com/currantlabs/gatt/peripheral_darwin.go:253 +0x7f
exit status 2
Exit code 1
```

broady, 2016-07-18:
```
$ go version
go version devel +29ed5da Wed Jul 13 21:18:19 2016 +0000 darwin/amd64
```

deadprogram, 2016-07-19:
You have to disable the cgo check by using the flag `GODEBUG=cgocheck=0` flag when running on OSX, as mentioned here: https://github.com/paypal/gatt/issues/76

For example, use:

```
GODEBUG=cgocheck=0 go run rover/ollie/1-color.go BB-EFCC
```

deadprogram, 2016-12-21:
The docs are now updated with this info. Thanks, everyone.

closed 2016-12-21 by deadprogram