hybridgroup/gobot #301: Platform request: ESP8266

AndreyS1, 2016-08-15:
Would be great if gobot could support this chip as well.

icecrasher321, 2016-08-16:
Agree with him. With its brilliant and cheap iot capabilities, the ESP8266 would be a great addition.

deadprogram, 2016-09-06:
This should be totally doable, by running StandardFirmataWiFi on the ESP8266, and then by passing a struct whihc is a `net/UDPConn` instead of a `tarm/goserial` in the `NewFirmataAdaptor` constructor, since both support `io.ReadWriteCloser`.

olliephillips, 2016-09-23:
I have no idea what's involved here. But I use Espruino on ESP8266. Being able to use Go would be amazing. It's also worth mentioning ESP3212, which has started to ship, could be more significant in terms of impact than even the ESP8266 was/is

deadprogram, 2016-10-04:
Hi @olliephillips running Golang directly on MCUs is not currently possible. However, in similar fashion to how Espruino is a stripped-down JavaScript, there are a couple of projects trying to get a Golang-like syntax running on the MCU itself. Check out https://github.com/ziutek/emgo

olliephillips, 2016-10-04:
@deadprogram thanks I'll check out that link. In terms of gobot.io can we currently treat the ESP8266 as an arduino platform? This was where my thinking started out.

deadprogram, 2016-10-04:
@olliephillips yes, it can be treated as an Arduino if you are running the Firmata firmware on it. However, we will still need a "serial over TCP/UDP" adaptor for Gobot.

deadprogram, 2017-01-02:
Hi everyone. Just commited some work to the `dev` branch with a preliminary implementation of `TCPAdaptor` for Firmata. Check out https://github.com/hybridgroup/gobot/blob/dev/examples/wifi_firmata_blink.go which works with my NodeMCU 1.0 ESP8266 running WifiFirmata.

deadprogram, 2017-01-09:
Experimental support for this was just released as part of v1.1.0. Now closing.

closed 2017-01-09 by deadprogram