hybridgroup/gobot #302: Is it possible to use gobot libraries without all the gobot/robot initialization.

ericho, 2016-08-18:
I'm wondering if it's possible to have a program like this 

```go
package main

import (
    "time"

    "github.com/hybridgroup/gobot/platforms/gpio"
    "github.com/hybridgroup/gobot/platforms/intel-iot/edison"
)

func main() {
    e := edison.NewEdisonAdaptor("edison")
    led := gpio.NewLedDriver(e, "led", "13")
    for {
        led.Toggle()
        time.Sleep(1)
    }
}
```

That could be part of another program. For instance if I want to read values from a sensor and send it to a database or another process. I think that I can achieve this by using a channel but the idea would be to have a simple way to get the ports/sensors data, something like the mraa library. 

Is there a way to have such functionality on gobot? 

Thanks

trevrosen, 2016-08-20:
Try something like this, but with the Edison adaptor and let me know what happens.

https://github.com/hybridgroup/gobot/blob/dev/examples/beaglebone_basic_direct_pin.go

trevrosen, 2016-08-20:
You're basically on the right track, you just need to `Connect()` the adaptor.

ericho, 2016-08-23:
Awesome! The `connect()` was the thing I needed. Now this code works

```go
package main

import (
	"time"
	"github.com/hybridgroup/gobot/platforms/gpio"
	"github.com/hybridgroup/gobot/platforms/intel-iot/edison"
)

func main() {
	e := edison.NewEdisonAdaptor("edison")
	led := gpio.NewLedDriver(e, "led", "13")
	e.Connect()
	for {
		led.Toggle()
		time.Sleep(1000 * time.Millisecond)
	}
}
```

Thanks :)

closed 2016-08-23 by ericho

deadprogram, 2016-08-23:
One other small thing worth mentioning. You should also call `Start()` on the driver, to ensure that it is properly initialized:

```
package main

import (
    "time"
    "github.com/hybridgroup/gobot/platforms/gpio"
    "github.com/hybridgroup/gobot/platforms/intel-iot/edison"
)

func main() {
    e := edison.NewEdisonAdaptor("edison")
    led := gpio.NewLedDriver(e, "led", "13")
    e.Connect()
    led.Start()
    for {
        led.Toggle()
        time.Sleep(1000 * time.Millisecond)
    }
}
```

deadprogram, 2016-08-23:
We should update the docs with this info, so people realize they can use Gobot as a set of Golang device libraries, not just use the entire framework.

trevrosen, 2016-08-23:
@deadprogram if you want to make an issue and assign it to me w/ some details about what you'd like to see and what platforms to cover, I can take on that docs update.

deadprogram, 2016-08-25:
Hi @trevrosen I just added https://github.com/hybridgroup/gobot-site/issues/48 to discuss this.