hybridgroup/gobot #304: Cannot read from unknown characteristic rolling spider via the BLE support from latest gobot

deadjoe, 2016-08-27:
Hi , after long waiting finally got the ble support to my device(rolling spider). 

Just gave it a try and failed to

deadjoe, 2016-08-27:
Just to confirm that the central to device are working well, i tried the cylon.js via https://github.com/ChrisTheBaron/cylon-rolling-spider and it does working perfect.

```
"use strict";

var Cylon = require("cylon");

Cylon.robot({
       	connections: {
       		/*
       		 * Remember to change the uuid to your own devices uuid
       		 */
       		spider: { adaptor: "rolling-spider", uuid: "8b2f8032290143e18fc7c426619632e8" }
       	},

       	devices: {
       		drone: { driver: "rolling-spider" }
       	},

       	work: function (my) {
       		my.drone.wheelOn();
       		my.drone.flatTrim();
       		my.drone.takeOff();

       		after(5000, function () {
       			my.drone.land();
       		});
       	}
}).start();
```

deadprogram, 2016-08-31:
Wow @deadjoe that is one incredibly well entered Github issue. Let me look into that on my OSX machine, and I will get back to you.

deadjoe, 2016-09-12:
Any luck  or clue?

deadprogram, 2016-09-14:
Hi @deadjoe I do not think that the Parrot Minidrone supports those particular characteristics. Did you try running https://github.com/hybridgroup/gobot/blob/master/examples/minidrone.go by any chance?

deadjoe, 2016-09-22:
Hi @deadprogram , below are the result when running minidrone.go

```
 examples git:(master) ✗ go run minidrone.go 8b2f8032290143e18fc7c426619632e8
2016/09/22 21:33:38 Initializing Robot bleBot ...
2016/09/22 21:33:38 Initializing connections...
2016/09/22 21:33:38 Initializing connection ble ...
2016/09/22 21:33:38 Initializing devices...
2016/09/22 21:33:38 Initializing device drone ...
2016/09/22 21:33:38 Starting Robot bleBot ...
2016/09/22 21:33:38 Starting connections...
2016/09/22 21:33:38 Starting connection ble...
State: PoweredOn
scanning...
2016/09/22 21:33:39 Unhandled event: xpc.Dict{"kCBMsgId":81, "kCBMsgArgs":xpc.Dict{"kCBMsgArgConnectionInterval":22, "kCBMsgArgConnectionLatency":0, "kCBMsgArgSupervisionTimeout":500, "kCBMsgArgDeviceUUID":xpc.UUID{0x8b, 0x2f, 0x80, 0x32, 0x29, 0x1, 0x43, 0xe1, 0x8f, 0xc7, 0xc4, 0x26, 0x61, 0x96, 0x32, 0xe8}}}
2016/09/22 21:33:39 Unhandled event: xpc.Dict{"kCBMsgId":53, "kCBMsgArgs":xpc.Dict{"kCBMsgArgDeviceUUID":xpc.UUID{0x8b, 0x2f, 0x80, 0x32, 0x29, 0x1, 0x43, 0xe1, 0x8f, 0xc7, 0xc4, 0x26, 0x61, 0x96, 0x32, 0xe8}, "kCBMsgArgATTMTU":104}}

Connected Peripheral ID:8b2f8032290143e18fc7c426619632e8, NAME:()
Failed to set MTU, err: Not implemented
2016/09/22 21:33:40 Unhandled event: xpc.Dict{"kCBMsgId":81, "kCBMsgArgs":xpc.Dict{"kCBMsgArgConnectionInterval":18, "kCBMsgArgConnectionLatency":0, "kCBMsgArgSupervisionTimeout":500, "kCBMsgArgDeviceUUID":xpc.UUID{0x8b, 0x2f, 0x80, 0x32, 0x29, 0x1, 0x43, 0xe1, 0x8f, 0xc7, 0xc4, 0x26, 0x61, 0x96, 0x32, 0xe8}}}
2016/09/22 21:33:41 Starting devices...
2016/09/22 21:33:41 Starting device drone...
2016/09/22 21:33:41 Starting work...
[4 1 2 3 0 0]
panic: runtime error: cgo argument has Go pointer to Go pointer

goroutine 35 [running]:
panic(0x40ea540, 0xc42011c450)
	/usr/local/go/src/runtime/panic.go:500 +0x1a1
github.com/currantlabs/gatt/xpc._cgoCheckPointer0(0xc42005a104, 0x0, 0x0, 0x0, 0x40dd601)
	??:0 +0x59
github.com/currantlabs/gatt/xpc.valueToXpc(0x40dd6e0, 0xc42011a5e0, 0x97, 0x40dd6e0)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:254 +0x3ff
github.com/currantlabs/gatt/xpc.valueToXpc(0x40e7f80, 0xc42011c440, 0x94, 0x40df300)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:269 +0x68b
github.com/currantlabs/gatt/xpc.valueToXpc(0x40fe180, 0xc420113890, 0x15, 0x40fe180)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:239 +0x8b2
github.com/currantlabs/gatt/xpc.valueToXpc(0x40e7f80, 0xc42011c380, 0x94, 0x40df300)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:269 +0x68b
github.com/currantlabs/gatt/xpc.valueToXpc(0x40fe180, 0xc4201138c0, 0x15, 0xc4201138c0)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:239 +0x8b2
github.com/currantlabs/gatt/xpc.goToXpc(0x40fe180, 0xc4201138c0, 0x0)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:213 +0x57
github.com/currantlabs/gatt/xpc.(*XPC).Send(0xc4200a0030, 0x40fe180, 0xc4201138c0, 0xc4200d6e00)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/xpc/xpc_darwin.go:23 +0x85
github.com/currantlabs/gatt.(*device).sendCBMsg(0xc4200a0000, 0x42, 0xc420113890)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/device_darwin.go:509 +0x15b
github.com/currantlabs/gatt.(*peripheral).loop.func1(0xc4200a0200, 0xc4200cc3c0)
	/Users/joe/goproj/src/github.com/currantlabs/gatt/peripheral_darwin.go:243 +0x119
created by github.com/currantlabs/gatt.(*peripheral).loop
	/Users/joe/goproj/src/github.com/currantlabs/gatt/peripheral_darwin.go:253 +0x7f
exit status 2
```

deadjoe, 2016-09-22:
and corresponding code of /Users/joe/goproj/src/github.com/currantlabs/gatt/peripheral_darwin.go:253 is 

```
236 func (p *peripheral) loop() {
237         rspc := make(chan message)
238
239         go func() {
240                 for {
241                         select {
242                         case req := <-p.reqc:
243                                 p.d.sendCBMsg(req.id, req.args)
244                                 if req.rspc == nil {
245                                         break
246                                 }
247                                 m := <-rspc
248                                 req.rspc <- m.args
249                         case <-p.quitc:
250                                 return
251                         }
252                 }
253         }()
```

deadprogram, 2016-09-25:
Hi, @deadjoe 

Just noticed you have the `panic: runtime error: cgo argument has Go pointer to Go pointer` error as discussed here: https://github.com/hybridgroup/gobot/issues/298

Please try running your code like this:

```
GODEBUG=cgocheck=0 go run minidrone.go 8b2f8032290143e18fc7c426619632e8
```

deadprogram, 2016-10-21:
Hi @deadjoe did you ever have a chance to try the command i suggested?

juliancheal, 2016-11-16:
I have a feeling this is related to https://github.com/currantlabs/ble/issues/16 in either `currantlabs/ble` or `currantlabs/gatt`

deadprogram, 2016-12-21:
The docs are now updated with this info. Thanks, everyone.

closed 2016-12-21 by deadprogram