hybridgroup/gobot #307: Change new adaptor/new driver function signatures to not require name

deadprogram, 2016-08-31:
This is derived from an email conversation I had with @mattetti thanks Matt!

It started me thinking about the naming thing. Seems like a little more separation of concerns might be useful to better accommodate devs who want to use Gobot as library not the full framework.

Specifically, I think that devs who care about a specific name to be used by the API can easily define that themselves, and everyone else will be satisfied with an auto-generated name, or not care about the internal name.

So what is currently this:

```go
e := edison.NewEdisonAdaptor("edison")
led := gpio.NewLedDriver(e, "led", "13")
button := gpio.NewButtonDriver(e, "button", "5")
```

would become:

```go
e := edison.NewEdisonAdaptor()
led := gpio.NewLedDriver(e, "13")
button := gpio.NewButtonDriver(e, "5")
```

If you wanted a specific name, you would then just do this:

```go
e := edison.NewEdisonAdaptor()
e.Name = "board"
led := gpio.NewLedDriver(e, "13")
led.Name = "flashlight"
button := gpio.NewButtonDriver(e, "5")
button.Name = "on-switch"
```

I think this will simplify most people use cases, however it would mean a breaking change to the new Adaptor and new Driver functions.

Feedback, please!

trevrosen, 2016-09-02:
👍 this is much more idiomatic and clear. Since it's a breaking change, would you want to take the version to `1.0.0` here? That's what semver compliance would require. That being said, there are probably enough people NOT doing this yet that you might be able to get away with just a minor version bump (i.e. to `0.13.1`)

agocs, 2016-09-02:
Would it make sense to allow currying `NewAdaptor()` and `Name`, e.g.

```
e := edison.NewAdaptor().Name("board")
```

deadprogram, 2016-09-03:
@trevrosen I was thinking exactly the same thing.

@agocs given that currying is not a common Golang idiom, I am very hesitant to add this.

agocs, 2016-09-03:
@deadprogram It wouldn't have to disallow the other method. E.g.

```
func (a *Adaptor) Name(name string) *Adapter {}
```

would both mutate the adaptor on which it is a method, and would return a reference to the mutated adaptor such that

```
e := edison.NewAdaptor()
e.Name("Bob")
```

is equivalent to

```
e := edison.NewAdaptor().Name("Bob")
```

Nevertheless, I understand why you might not want to add an affordance for currying.

trevrosen, 2016-09-06:
Yeah I'd also be hesitant to use currying here. I feel like porting the existing behavior of passing in the name as the constructor's only argument keeps it simple and honors the principle of least surprise.

deadprogram, 2016-09-06:
Well-thought out feedback, thanks.

@trevrosen adding `SetName` to the interface, combined with a clearly-defined auto-naming, seems like a nicer way to go. The common case where there are params to the constructor that are significant such as pin numbers or a port, are more interesting to consider than the simpler Edison case. For example, if we just go with "default" names:

```go
a := firmata.NewFirmataAdaptor("firmata", "/dev/ttyACM0")
led := gpio.NewLedDriver(a, "led", "13")
```

looks a lot better to me as:

```go
a := firmata.NewAdaptor("/dev/ttyACM0") // Name() == "firmata"
led := gpio.NewLedDriver(a, "13") // Name() == "led"
```

If we do care about specific names to be used by external APIs:

```go
a := firmata.NewFirmataAdaptor("arduino", "/dev/ttyACM0")
led := gpio.NewLedDriver(a, "blinker", "13")
```

becomes

```go
a := firmata.NewAdaptor("/dev/ttyACM0")
a.SetName("arduino")
led := gpio.NewLedDriver(a, "13")
led.SetName("blinker")
```

I do appreciate the consideration of reducing the breakage, but since it would change anyhow, seems like getting the work out of the way all at once would be kinder.

deadprogram, 2016-09-25:
This is the branch where my WIP on this can be found: https://github.com/hybridgroup/gobot/tree/feature/new-adaptor-driver-signatures

deadprogram, 2016-10-10:
This has now been merged into the `dev` branch, thanks everyone for all the great feedback!

closed 2016-10-10 by deadprogram