hybridgroup/gobot #310: I²C mcp23017 pin input mode

th002, 2016-09-06:
How to setting mcp23017 pin input mode?

deadprogram, 2016-09-06:
Hi @th002 looks to me that https://github.com/hybridgroup/gobot/blob/dev/platforms/i2c/mcp23017_driver.go#L160 is missing some code like https://github.com/hybridgroup/gobot/blob/dev/platforms/i2c/mcp23017_driver.go#L147-L150 to set to input mode.

I do not have one of these handy to test out with at the moment, but a PR with this would be gladly accepted.

th002, 2016-09-06:
I had add function PinMod
```
func (m *MCP23017Driver) PinMode(pin, val uint8, portStr string) (err error) {
	selectedPort := m.getPort(portStr)
	// Set IODIR register bit for given pin to an output/input.
	if err = m.write(selectedPort.IODIR, uint8(pin), val); err != nil {
		return
	}
	return
}
```

deadprogram, 2016-09-06:
That looks like a good approach.

closed 2016-09-07 by th002

deadprogram, 2016-09-07:
Hi @th002 can you submit a PR with this fix? I'm sure all users of this expander would appreciate it.

Thanks!