hybridgroup/gobot #311: raspberry pi 3 support SMBus block read

th002, 2016-09-07:
raspberry pi 3 support SMBus block read, but gobot/sysfs/i2c_device.go L92

deadprogram, 2016-09-07:
If the platform does not support block reads, it performs the file-based read https://github.com/hybridgroup/gobot/blob/dev/sysfs/i2c_device.go#L94 

Otherwise, it continues executing the `Read()` using the block-based read https://github.com/hybridgroup/gobot/blob/dev/sysfs/i2c_device.go#L97-L119

Hope that helps!

th002, 2016-09-07:
The data from file-based read is difference of the data from SMBus block-base read

deadprogram, 2016-09-07:
Hi @th002 please provide some more specific details about the differences, and then we can come up with a plan to accommodate them. Sounds good?

th002, 2016-09-09:
```
package main

import (
	"fmt"
	"os"
	"syscall"
	"unsafe"
)

const (
	I2C_SLAVE = 0x0703
	I2C_SMBUS = 0x0720

	I2C_SMBUS_READ           = 1
	I2C_SMBUS_I2C_BLOCK_DATA = 8
	// Adapter functionality
	I2C_FUNCS = 0x0705
)

type i2cSmbusIoctlData struct {
	readWrite byte
	command   byte
	size      uint32
	data      uintptr
}

func main() {
	f, err := os.OpenFile("/dev/i2c-1", os.O_RDWR, os.ModeExclusive)
	if err != nil {
		fmt.Println("open file error====>", err)
		return
	}

	var funcs uint64
	_, _, errno := syscall.Syscall(
		syscall.SYS_IOCTL,
		f.Fd(),
		I2C_FUNCS,
		uintptr(unsafe.Pointer(&funcs)),
	)
	if errno != 0 {
		fmt.Printf("Querying functionality failed with syscall.Errno %v\n", errno)
		return
	}

	_, _, errno = syscall.Syscall(
		syscall.SYS_IOCTL,
		f.Fd(),
		I2C_SLAVE,
		uintptr(byte(0x20)),
	)

	if errno != 0 {
		fmt.Printf("Setting address failed with syscall.Errno %v\n", errno)
		return
	}

	//SMBus block read mcp23017 pin 0 port A
	b := make([]byte, 19)
	data := make([]byte, len(b)+1)
	data[0] = byte(len(b))

	smbus := &i2cSmbusIoctlData{
		readWrite: I2C_SMBUS_READ,
		command:   0,
		size:      I2C_SMBUS_I2C_BLOCK_DATA,
		data:      uintptr(unsafe.Pointer(&data[0])),
	}
	_, _, errno = syscall.Syscall(
		syscall.SYS_IOCTL,
		f.Fd(),
		I2C_SMBUS,
		uintptr(unsafe.Pointer(smbus)),
	)

	if errno != 0 {
		fmt.Printf("Read failed with syscall.Errno %v\n", errno)
		return
	}
	copy(b, data[1:])
	fmt.Println("read from SMBus data==========>", b)
	f.Close()

	f, err = os.OpenFile("/dev/i2c-1", os.O_RDWR, os.ModeExclusive)
	if err != nil {
		fmt.Println("open file error====>", err)
		return
	}
	defer f.Close()

	_, _, errno = syscall.Syscall(
		syscall.SYS_IOCTL,
		f.Fd(),
		I2C_FUNCS,
		uintptr(unsafe.Pointer(&funcs)),
	)

	if errno != 0 {
		fmt.Printf("Querying functionality failed with syscall.Errno %v\n", errno)
		return
	}

	_, _, errno = syscall.Syscall(
		syscall.SYS_IOCTL,
		f.Fd(),
		I2C_SLAVE,
		uintptr(byte(0x20)),
	)

	if errno != 0 {
		fmt.Printf("Setting address failed with syscall.Errno %v\n", errno)
		return
	}

        //file-base read mcp23017 pin 0 port A
	data2 := make([]byte, 19)
	_, err = f.Read(data2)
	if err != nil {
		fmt.Println("file-based read error====>", err)
		return
	}
	fmt.Println("read from file-base data=======>", data2)
}
```
* This result is the first run this code:
```
    read from SMBus data==========> [255 255 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 255]
    read from file-base data=======> [127 0 0 255 255 0 0 0 0 0 0 0 0 0 0 1 0 0 0]
```

* This result is the second run this code:
```
   read from SMBus data==========> [255 255 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 3]
   read from file-base data=======> [0 0 0 255 255 0 0 0 0 0 0 0 0 0 0 1 0 0 0]
```

* This result is the third run this code:
```
   read from SMBus data==========> [255 255 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 3]
   read from file-base data=======> [0 0 0 255 255 0 0 0 0 0 0 0 0 0 0 1 0 0 0]
```

* This result is the fourth run this code:
```
   read from SMBus data==========> [255 255 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 255]
   read from file-base data=======> [255 0 0 255 255 0 0 0 0 0 0 0 0 0 0 1 0 0 0]
```

Platform is raspberry pi 3 B

closed 2016-09-20 by th002

deadprogram, 2016-09-20:
Hi @th002 did you mean to close this on purpose?

th002, 2016-09-20:
Yeah,Raspberry pi 3 is not support SMBus block read.
```
 smbus := &i2cSmbusIoctlData{
        readWrite: I2C_SMBUS_READ,
        command:   0,
        size:      I2C_SMBUS_I2C_BLOCK_DATA,
        data:      uintptr(unsafe.Pointer(&data[0])),
    }
```
This code is i2c_smbus_read_i2c_block_data?