hybridgroup/gobot #313: mqtt from git.eclipse.org is not accepted any more.

johanhenselmans, 2016-09-12:
Doing : go get -d -u github.com/hybridgroup/gobot/...
resulted in:
package git.eclipse.org/gitroot/paho/org.eclipse.paho.mqtt.golang.git: cannot download, git://git.eclipse.org/gitroot/paho/org.eclipse.paho.mqtt.golang uses insecure protocol

changing 
import (
	"git.eclipse.org/gitroot/paho/org.eclipse.paho.mqtt.golang.git"
	"github.com/hybridgroup/gobot"
)

to: 
import (
	"github.com/eclipse/paho.mqtt.golang"
	"github.com/hybridgroup/gobot"
)

Solved this.

deadprogram, 2016-09-12:
Hi @johanhenselmans thanks for pointing this out. A PR with this fix would be gratefully accepted :)

deadprogram, 2016-10-03:
Hi @johanhenselmans this has been fixed as part of https://github.com/hybridgroup/gobot/commit/17cc57d9d1844a3fc97234fd8ae2304552cb9e12

I will leave this issue open until that code is released.

deadprogram, 2016-10-05:
This code is now in PR https://github.com/hybridgroup/gobot/pull/316

deadprogram, 2016-10-10:
This has now been merged into the `dev` branch. Please re-open if needed.

closed 2016-10-10 by deadprogram