hybridgroup/gobot #314: Support for kuman relay boards?

DaBlitzStein, 2016-09-14:
Hello,
I understad Kuman relay boards are not supported (https://www.amazon.es/gp/product/B014L10Q52/ref=s9_simh_gw_g147_i2_r?pf_rd_m=A1AT7YVPFBWXBL&pf_rd_s=&pf_rd_r=TX9715MK7HA329ZJWR8V&pf_rd_t=36701&pf_rd_p=6d8f0343-6251-45c8-9899-ea554850c331&pf_rd_i=desktop)

If i am wrong and they are supported, how can use it (Raspberry PI 3B+)

deadprogram, 2016-09-14:
Hola, @DaBlitzStein 

Looks like that board is basically 8 relays on a single board. You should be able to use the Gobot `Relay` driver with it, just by connecting the various Raspi pins to the `IN1`, `IN2` pins on the Kuman board.

For example, this should work with the Raspi pins 32-40 conneccted to IN1 thru IN8 on the Kuman board:

```go
package main

import (
	"time"

	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/platforms/raspi"
  "github.com/hybridgroup/gobot/platforms/gpio"
)

func main() {
	gbot := gobot.NewGobot()

	r := raspi.NewRaspiAdaptor("raspi")
	relay1 := gpio.NewRelayDriver(r, "relay1", "33")
        relay2 := gpio.NewRelayDriver(r, "relay2", "34")
        relay3 := gpio.NewRelayDriver(r, "relay3", "35")
        relay4 := gpio.NewRelayDriver(r, "relay4", "36")
        relay5 := gpio.NewRelayDriver(r, "relay5", "37")
        relay6 := gpio.NewRelayDriver(r, "relay6", "38")
        relay7 := gpio.NewRelayDriver(r, "relay7", "39")
        relay8 := gpio.NewRelayDriver(r, "relay8", "40")

	work := func() {
		gobot.Every(3*time.Second, func() {
			relay1.Toggle()
                        relay2.Toggle()
                        relay3.Toggle()
                        relay4.Toggle()
                        relay5.Toggle()
                        relay6.Toggle()
                        relay7.Toggle()
                        relay8.Toggle()
		})
	}

	robot := gobot.NewRobot("kuman",
		[]gobot.Connection{r},
		[]gobot.Device{relay1, relay2, relay3, relay4, relay5, relay6, relay7, relay8},
		work,
	)

	gbot.AddRobot(robot)

	gbot.Start()
}
```

DaBlitzStein, 2016-09-14:
Thank you, I will try. I tell you. If not, we will try to develop the driver.

deadprogram, 2016-09-17:
Hi, @DaBlitzStein I'm going to close this issue. Please reopen if you need further assist.

closed 2016-09-17 by deadprogram

DaBlitzStein, 2016-09-17:
It worked perfectly, only the pinout have to be changed due the gnd pins. After this everything ok. Thank you!